olikraus/u8g2 #1268: Image shown halved and mirrored

Rothpe, 2020-08-31:
Hi,

i hope you can help me.
Display is an OLED 0,49 ".
I used U8G2 Lib and draw a Bitmap with "u8g2.drawXBMP(25, 0, 30, 26, myBitmap);"
But Image is display halved and mirrored.
Right halve image is display on left side and left halve image is displayed on right side.
Constructor is: U8G2_SSD1306_64X32_1F_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

Thanks in advance for your support.

Best regards

Peter

olikraus, 2020-08-31:
What about the hello world example? Is this correct?

Rothpe, 2020-09-01:
Yes there are 6 lines display correctly. Example works fine

Rothpe, 2020-09-01:
[https://www.directupload.net/file/d/5929/lttrm6zk_jpg.htm](url)
Pls find attached link for image shown 0,49 " Display

olikraus, 2020-09-01:
Not sure what I can say here. How does the code look like? What is the content of the XBM file? 
I mean XBM has been working for many users, so probably you need to exclude all other sources of errors, especially if you say, that the examples are working correctly.

Rothpe, 2020-09-01:
`// Display - https://pmdway.com/collections/oled-displays/products/0-49-64-x-32-white-graphic-oled-i2c
// Guide - https://pmdway.com/blogs/product-guides-for-arduino/tutorial-using-the-0-49-64-x-32-graphic-i2c-oled-display-with-arduino

#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

//U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/U8X8_PIN_NONE);
//U8G2_SSD1306_64X32_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
U8G2_SSD1306_64X32_1F_F_HW_I2C u8g2(U8G2_R0, /* reset=*/U8X8_PIN_NONE);

const unsigned char myBitmap[] PROGMEM = {
    B00000000, B00000000, B00000000, B00000000,
    B00000000, B00000011, B00000000, B00000000,
    B00000000, B00000111, B10000000, B00000000,
    B00000000, B00000111, B10000000, B00000000,
    B00000000, B00001111, B11000000, B00000000,
    B00000000, B00001111, B11000000, B00000000,
    B00000000, B00011100, B11100000, B00000000,
    B00000000, B00111100, B11100000, B00000000,
    B00000000, B00111000, B01110000, B00000000,
    B00000000, B01111000, B01111000, B00000000,
    B00000000, B01110000, B00111000, B00000000,
    B00000000, B11100000, B00011100, B00000000,
    B00000000, B11101110, B11011100, B00000000,
    B00000001, B11000111, B00001110, B00000000,
    B00000011, B11011111, B00001111, B00000000,
    B00000011, B10011111, B00000111, B00000000,
    B00000111, B10011110, B00000111, B10000000,
    B00000111, B00010110, B00000011, B10000000,
    B00001110, B00010010, B10000001, B11000000,
    B00001110, B00010110, B00110001, B11000000,
    B00011100, B00110100, B01111000, B11100000,
    B00111100, B00000000, B11111100, B11110000,
    B00111000, B00000000, B00000000, B01110000,
    B01111111, B11111111, B11111111, B11111000,
    B01111111, B11111111, B11111111, B11111000,
    B00000000, B00000000, B00000000, B00000000};

void setup()
{
  u8g2.begin();
}

// fonts https://github.com/olikraus/u8g2/wiki/fntlistall#4-pixel-height

void loop()
{
  u8g2.clearBuffer();                  // clear the internal memory
  u8g2.setFont(u8g2_font_u8glib_4_tf); // choose a suitable font
  u8g2.drawStr(0, 5, "Hello,");        // write something to the internal memory
  u8g2.drawStr(0, 10, "World...");
  u8g2.drawStr(0, 15, "I'm tiny...");
  u8g2.drawStr(0, 20, "So tiny!");
  u8g2.drawStr(0, 25, "However you can");
  u8g2.drawStr(0, 30, "have six lines");
  u8g2.sendBuffer(); // transfer internal memory to the display
  delay(3000);

  /*u8g2.clearBuffer();
  
  for (int x = 0; x <= 50; x++)
  {
    for (int y = 0; y <= 29; y++)
    {
      u8g2.drawPixel(x, y);
      u8g2.sendBuffer();
    }
  }

  delay(2000);*/
  u8g2.clearBuffer();
  u8g2.setDrawColor(1);
  u8g2.setBitmapMode(0);
  u8g2.drawXBMP(5, 5, 30, 26, myBitmap);
  u8g2.sendBuffer();

  /*u8g2.clearBuffer();               // clear the internal memory
  u8g2.setFont(u8g2_font_t0_11_tf); // choose a suitable font
  u8g2.drawStr(0, 10, "Hello,");    // write something to the internal memory
  u8g2.drawStr(0, 20, "World...");
  u8g2.drawStr(0, 30, "I'm tiny...");
  u8g2.sendBuffer(); // transfer internal memory to the display
  delay(1000);

  u8g2.clearBuffer();                   // clear the internal memory
  u8g2.setFont(u8g2_font_tenstamps_mf); // choose a suitable font
  u8g2.drawStr(0, 12, "ABCD");          // write something to the internal memory
  u8g2.drawStr(0, 30, "1234");
  u8g2.sendBuffer(); // transfer internal memory to the display
  delay(1000);

  for (int a = 999; a >= 0; --a)
  {
    u8g2.clearBuffer();               // clear the internal memory
    u8g2.setFont(u8g2_font_inb24_mr); // choose a suitable font
    u8g2.setCursor(0, 24);
    u8g2.print(a);
    a = a - 47;
    u8g2.sendBuffer(); // transfer internal memory to the display
    delay(100);
  }*/
  delay(20000);
}`

Rothpe, 2020-09-01:
Hi
at first thank you for your quick support.
Perhaps you can check code.
I found a thread inArduino Forum and he had same problem.
But there was no solution
I tested a lot of functionality with this display thats the reason why some code is commented out.

Best regards

Peter

olikraus, 2020-09-01:
Unfortunately you reversed the bit order of the bitmap. I mean XBM is an official format, supported by many drawing tools and converters (https://en.wikipedia.org/wiki/X_BitMap), so the the leftmost pixel in a byte is rightmost pixel on the display. However in your manual written example, you assume, that the leftmost pixel is also the left most pixel on the screen, which is not the case in the XBM format.

You  need to revert the bytes in your example, for example byte 2 in line 2 should not be B00000011 but B11000000 instead.
As a consequence you can not check the bitmap in the code. 

Also the description (halved and mirrored) is misleading, from my perspective there is no real word for this. ;-)

Rothpe, 2020-09-02:
Hi,

Thank you very much for the quick answer and the great tip.
Your support is very good and not something you can take for granted.
Hence my respect for it.
Thank you again for that.

Best wishes

Peter

closed 2020-09-02 by Rothpe

olikraus, 2020-09-02:
:)