olikraus/u8g2 #1270: u8g2 freezes on Arduino Nano when using PORTC

zzarko, 2020-09-03:
This very well may not be an u8g2 issue, so I apologize in advance if I misunderstood something and used u82g in a way that isn't intended...

Anyhow, I started to work on second version of my joystick tester (https://bitbucket.org/zzarko/joysticktester) a couple of weeks ago, this time with rotary encoder and OLED. After adding a lot of stuff, I managed to use almost all RAM and decided to change my data structures and to use direct port read/write instead of using digitalRead/Write and pinMode. But I hit a show-stopping issue when I tried to use PORTC together with u8g2 library. Here is the minimal code that illustrates the issue:

`// 1.3" OLED, connected on A4/A5
#include <U8g2lib.h>
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
const uint8_t *fontSmall = u8g2_font_profont10_mr; //u8g2_font_profont10_tr;

void setup() {
  // setup OLED
  u8g2.begin();
  u8g2.clearBuffer();
  u8g2.setFont(fontSmall);
  u8g2.setFontRefHeightExtendedText();
  u8g2.setDrawColor(1);
  u8g2.setFontPosTop();
  u8g2.setFontDirection(0);

  // setup pins
  DDRB = 0x10 | 0b00100000; // additional constant for LED
  PORTB = 0x1F;
  DDRC &= 0x11110000;
  DDRC |= 0x04;

  // *** this doesn't work ***
//  PORTC &= 0x11110000;
//  PORTC |= 0x0B;

  // *** this works ***
  digitalWrite(A0, HIGH);
  digitalWrite(A1, HIGH);
  digitalWrite(A2, LOW);
  digitalWrite(A3, HIGH);

}

void loop() {
  static boolean led = true;

  u8g2.clearBuffer();
  u8g2.setCursor(10,10);
  u8g2.print("PINB:");
  u8g2.print(PINB & 0b11011111, HEX); // exclude LED pin
  u8g2.setCursor(10,20);
  u8g2.print("PINC:");
  u8g2.print(PINC, HEX);
  u8g2.sendBuffer();

  // just to see if loop is running
  if (led) digitalWrite(13,1);
  else digitalWrite(13,0);
  led = !led;
}`

If I use digitalWrite to set A0-A3, everything works as expected. If I use PORTC to set them all at once, the code freezes right after reset of Arduino (LED sometimes flashes 1-2 times, but usually code stops after first run through loop). I do not see what I'm doing wrong with direct port access, it may be something obvious to more experienced Arduino coders.

As I need to be able to change pin meaning on the fly (20-50 times per second), I need to be able to do it after OLED setup (the values in the code are just one of many configurations that I have). I just do not know why digitalWrite access to pins work, while direct manipulation with PORTC doesn't.

I think that it has something to do with how u8g2 library works, as when I comment-out u8g2.sendBuffer() call, the LED on Arduino keeps flashing no matter which way I use to set A0-A3.

Again, sorry if this is something obvious, or if I didn't use u8g2 as intended...

olikraus, 2020-09-03:
Did you use external pull up resistors for I2C?

zzarko, 2020-09-03:
olikraus, I'm stupid and should not program when I didn't get enough sleep... I used 0x prefix for binary number in that one instance, I do not know why, of course it didn't work. I do not know what was I thinking, I'm terribly sorry for wasting your time. I do not have pull ups on I2C, but that wasn't the problem, obviously...

zzarko, 2020-09-03:
Also, please close this non-issue, as I was right in my first sentence...

olikraus, 2020-09-03:
No problem ;)

closed 2020-09-03 by olikraus