olikraus/u8g2 #1271: Huge lag with Seeed 128x128 display - u8g2 or u8x8 (CAN BUS)

simoniz101, 2020-09-03:
Hi,

I am trying to display my car's data on the Seeed Grove 1.12" v2 OLED display:
https://wiki.seeedstudio.com/Grove-OLED_Display_1.12inch/

My controller is an integrated CAN BUS board...which is the same as a Leonardo.
https://www.seeedstudio.com/CANBed-Arduino-CAN-BUS-Development-Kit-Atmega32U4-with-MCP2515-and-MCP2551-p-4365.html

My code is working perfectly when I use the Serial monitor. All the data is showing perfectly with the serial.print function, and is refreshing very fast. Even if I use the SeeedOled code, the display is updating very fast. However, there is no way to get bigger text than 1 line height with the Seeed code....so I want to use the u8g2 or u8x8 fonts.

My problem: The data shown in the display is updated wayyyy too slowly. The numbers are updated every 1-5 seconds even if I only try to display one parameter (such as RPM let's say). The RPM on a car is constantly changing so...I'm not even close.
I thought the u8g2 was maybe the cause and changed the program to accomodate u8x8. It got slightly faster, but still the display is refreshing ridiculously slow.

Here is my code:

```
#include <Arduino.h>

#include <U8x8lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#include <Wire.h>
#include <mcp_can.h>

U8X8_SH1107_SEEED_128X128_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);

MCP_CAN CAN(17);          //set SPI Chip Select to pin 17

unsigned char len = 0;
unsigned char buf[8];
unsigned int canID;
unsigned int boutonpage;

unsigned int rpm;
unsigned int man_pressure;
unsigned int th_position;
unsigned int fuel_press;
unsigned int speedo;
unsigned int gear;
unsigned int voltage;
unsigned int coolant_temp;
unsigned int air_temp;
unsigned int fuel_cons;
unsigned int economy;

void setup()
{
  Serial.begin(115200);   //to communicate with Serial monitor
  Wire.begin();
  u8x8.begin();
  u8x8.setPowerSave(0);
    

//tries to initialize, if failed --> it will loop here for ever
START_INIT:

    if(CAN_OK == CAN.begin(CAN_1000KBPS))      //setting CAN baud rate to 1000Kbps
    {
        Serial.println("CAN BUS Shield init ok!");
    }
    else
    {
        Serial.println("CAN BUS Shield init fail");
        Serial.println("Init CAN BUS Shield again");
        delay(100);
        goto START_INIT;
    }
}

void loop()
{

    if(CAN_MSGAVAIL == CAN.checkReceive())        //check if data coming
    {
        CAN.readMsgBuf(&len, buf);    //read data,  len: data length, buf: data buffer
        canID = CAN.getCanId();       //getting the ID of the incoming message

        if (canID == 0x360)            //reading only our beloved 0x360 message
        {
          //MAP
          man_pressure = word((buf[2]*256+buf[3]));
          //TPS
          th_position = word((buf[4]*256+buf[5]));
        }
          
        if (canID == 0x3E0)            //reading only our beloved 0x3E0 message
        {
          //Coolant temperature
          coolant_temp = word((buf[0]*256+buf[1]));
          }

        if (canID == 0x3E2)            //reading only our beloved 0x3E2 message
        {
          //Average fuel economy
          economy = word((buf[4]*256+buf[5]));
          }
     }

    u8x8.setFont(u8x8_font_inr21_2x4_r);
    u8x8.setCursor(0, 0);
    u8x8.print("MAP");
    u8x8.setFont(u8x8_font_inb33_3x6_n);
    u8x8.setCursor(0, 3);
    if (man_pressure <= 1013)
        {
          u8x8.print(((float(man_pressure)/10)-101.3)*0.2953, 1);
//          u8x8.setCursor(0, 80);
//          u8x8.setFont(u8x8_font_inr21_2x4_r);
//          u8x8.print(" inHG");
          }

  else
        {
          u8x8.print(((float(man_pressure)/10)-101.3)*0.145038, 1);
          u8x8.setCursor(0, 3);
          u8x8.setFont(u8x8_font_inr21_2x4_r);
          u8x8.print(" psi");
        }   
      u8x8.setFont(u8x8_font_7x14B_1x2_f);
      u8x8.setCursor(0, 9);
      u8x8.print("Coolant Temp");
      u8x8.setFont(u8x8_font_inr21_2x4_n);
      u8x8.setCursor(0, 11);
      u8x8.print(((float(coolant_temp)/10)-273.15), 1);
//      u8x8.setFont(u8x8_font_7x14B_1x2_f);
//      u8x8.print(" °C");
//    u8x8.setFont(u8x8_font_courB18_2x3_f);
//    u8x8.setCursor(0, 106);
//    u8x8.print("Fuel economy");
//    u8x8.setFont(u8x8_font_courB18_2x3_f);
//    u8x8.setCursor(0, 128);
//    u8x8.print((float(economy)/10), 1);
//    u8x8.print(" L/100km");
          
}
```

olikraus, 2020-09-03:
Maybe only update the display if you have received new CAN data.

simoniz101, 2020-09-03:
What do you mean update the display? I tried the "clearbuffer" / "sendbuffer" with u8g2 and also "refreshdisplay" with u8x8 with no luck...

olikraus, 2020-09-03:
No, I just meant if you could put the u8x8 into the if MSGAVAIL condition.

simoniz101, 2020-09-03:
Just tried it, no change. I counted almost 10 seconds before getting an updated value. CAN messages are always streaming so...

olikraus, 2020-09-03:
Did you try some of the u8x8 examples with this display? I don't think it's an u8g8 issue.

simoniz101, 2020-09-09:
Hi, just a comeback on the issue.
For those experiencing the same thing...it's not a u8x8 issue.

I was not filtering the CAN messages so the controller was too busy getting all the messages. I used the mask and filter functions to limit that and it improved the response a lot.

I also made only 1 data reading per 'page' in the display, so there is only one data to read over and over. I also used setBusClock(200000) since this is the maximum the display can handle. If you try to go above that, it crashes.

Finally, the display is showing acceptable refresh rate. I would be happier with faster, but I think I reached the limit...

closed 2022-03-04 by olikraus