olikraus/u8g2 #1272: garbled data with U8X8 on SED1520 compatible display

rowokii, 2020-09-04:
Hi,

I'm using Arduino IDE 1.8.13 with U8g2 version 2.27.6 installed.  The LCD is Displaytech 32122A, which has two NJU6460A display controller which is SED1520 compatible. (122x32 resolution)   Using a simple test code listed below, I can get the last character of each row printed on the first column of that row, like so:  

0junk data
1junk data
2 junk data
3 junk data
[image]

In fact, if I drawString "Hello world", on the first row, I can see each character print in order on the first row, first column, ending with the last "d", followed by the junk data for the rest of the row.   

The fact I can get the correct strings printed on the LCD seems to indicate I have correct wiring and enough of the data and command is making it to the LCD, but I can't figure out where the junk data is coming from.  I connected a scope to the signals and they don't look like electrical noise and when I hooked up a logic analyzer,  it really does look like the software is sending the garbled data to the lcd.   

I'm not really sure if I using the library incorrectly or missing some setup?    I'm really at a loss as to where to go from here.  Any help would be appreciated....

#include <Arduino.h>
#include <U8x8lib.h>
U8X8_SED1520_122X32 u8x8(0,1,2,3,4,5,6,7,11,9,10,8);
void setup(void) {
   u8x8.begin();
}
void loop(void) {
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  u8x8.drawString(0,0,"000");
  u8x8.drawString(1,1,"111");
  u8x8.drawString(2,2,"222");
  u8x8.drawString(3,3,"333")
}

rowokii, 2020-09-04:
Forgot to mention using Arduino Uno.

olikraus, 2020-09-04:
Will the u8g2 procedures work? I mean can you test one of the existing u8g2 examples?

rowokii, 2020-09-04:
Great suggestion!  U8g2 example worked.   So looks like it's u8x8 only problem, but at least I now have a workaround.   Thanks!

[image]

olikraus, 2020-09-04:
I crosschecked this: U8x8 is not supported. You can only use u8g2. It is a kind of a display issue here not a u8x8 problem as such. There are displays which do not support horizontal addressing. This is one of them.

rowokii, 2020-09-04:
OK.  Thanks for checking and thanks for the quick response!

olikraus, 2020-09-04:
:-)

closed 2020-09-04 by olikraus