olikraus/u8g2 #1273: Looking for some help with C Library SH1122 over SPI

ToastyStoemp, 2020-09-04:
Hello,

I've been working on a small keyboard project and was looking for some help with the implementation of the C Library for U8g2.

I've been looking at the examples and following them as closely as I can, however I can not get the code working at all. I've mainly based myself on the discussion found here: https://github.com/olikraus/u8g2/issues/175

I'm using an Atmega32u4 and was able to get the Arduino example sketches working without any problems. (page buffer)

However due to other software requirements I can not stick with Arduino. Therefor I've setup a C project where I've been trying to get the display to work from pure C code.

Here is my current main.c file:
```
/*
 * based on code by ing_mauro
 */ 

#define F_CPU 16000000UL
#include <avr/io.h>
#include <util/delay.h>
#include "u8g2.h"
#include "u8x8.h"

// PB1=9 SCLK 
// PB2=10 MOSI
// PF4=39 DC/A0
// PF1=40 CS 
// PF5=38 RESET 

u8g2_t u8g2;

uint8_t u8g2_gpio_and_delay_sh1122(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr){
	
	switch(msg){
		//Initialize SPI peripheral
		case U8X8_MSG_GPIO_AND_DELAY_INIT:
			break;
		//Function which implements a delay, arg_int contains the amount of ms
		case U8X8_MSG_DELAY_MILLI:
			//_delay_us(arg_int*1000UL);
			for(uint8_t i=arg_int; i > 0; i--) { // Modified here, AVR5
				_delay_ms(1);
			}
			
			break;
		case U8X8_MSG_DELAY_10MICRO:
			_delay_us(10);
			break;
		//Function to define the logic level of the clock line
		case U8X8_MSG_GPIO_SPI_CLOCK:
			if (arg_int == 0) {				// arg_int=0: Output low at SCLK pin
				DDRD |= (1<<DDB1);
				PORTD &= ~(1<<PORTB1);
			} else {						// arg_int=1: Input dir with pull up high for SCLK pin
				DDRD &= ~(1<<PORTB1);		// The same with the rest of pins
				PORTD |= (1<<PORTB1);
			}
			break;
		case U8X8_MSG_GPIO_SPI_DATA:
			if (arg_int == 0) {
				DDRD |= (1<<DDB2);
				PORTD &= ~(1<<PORTB2);
			} else {	
				DDRD &= ~(1<<PORTB2);
				PORTD |= (1<<PORTB2);
			}	
			break;	
		case U8X8_MSG_GPIO_CS:	
			if (arg_int == 0) {
				DDRB |= (1<<DDF1);
				PORTB &= ~(1<<PORTF1);
			} else {	
				DDRB &= ~(1<<PORTF1);
				PORTB |= (1<<PORTF1);
			}	
			break;		
		case U8X8_MSG_GPIO_DC:		
			if (arg_int == 0) {
				DDRD |= (1<<DDF4);
				PORTD &= ~(1<<PORTF4);
			} else {	
				DDRD &= ~(1<<PORTF4);
				PORTD |= (1<<PORTF4);
			}
			break;		
		case U8X8_MSG_GPIO_RESET:
			if (arg_int == 0) {
				DDRC |= (1<<DDF5);
				PORTC &= ~(1<<PORTF5);
			} else {	
				DDRC &= ~(1<<PORTF5);
				PORTC |= (1<<PORTF5);
			}
			break;
		default:
			return 0; //A message was received which is not implemented, return 0 to indicate an error
		}							
	return 1; // command processed successfully.
}

/*****************************************************************/
void displayInit(void){
	u8g2_Setup_sh1122_256x64_1(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_sh1122);
	u8g2_InitDisplay(&u8g2);		//Send initialization code to the display
	u8g2_SetPowerSave(&u8g2, 0);
}

int main(void) {
	displayInit();
	//u8g2.begin();
	while(1){
		// picture loop
		u8g2_ClearBuffer(&u8g2);
		u8g2_SetFont(&u8g2, u8g2_font_6x10_tf);
		u8g2_DrawStr(&u8g2, 0, 0, "Test");
		u8g2_SendBuffer(&u8g2);
	}
}
```

Any help would be appreciated. 

PS. The display module I got only supports SPI

olikraus, 2020-09-04:
I have created a step by step tutorial for AS7: https://github.com/olikraus/u8g2/wiki/u8g2as7

It will use the following code:
https://github.com/olikraus/u8g2/blob/master/sys/avr/as7/main.c

From a quick look, i see two issues here:

First, your gpio and delay procedure differs from the above mentioned main.c file. In your code you will change the data direction, where from my perspective only the output level should be modified. Maybe better takeover the code from the above mentioned main.c file

The second point is: You have used a page buffer constructor (' _1'). In such a case the picture loop is required, which would look like this:
``` C++
  u8g2_FirstPage(&u8g2);
  do {
    ...
  } while ( u8g2_NextPage(&u8g2) );
```

`u8g2_ClearBuffer(&u8g2)` and `u8g2_SendBuffer(&u8g2)` will only work in full buffer mode (constructor with `_f`).

ToastyStoemp, 2020-09-04:
Thank you for the swift reply,

I originally did follow the u8g2as7, great stuff to get me going with atmel studio, however I couldn't really figure out the pin mapping for it.
changing the pin numbers on https://github.com/olikraus/u8g2/blob/master/sys/avr/as7/main.c#L46 and others to be anything higher than 9 cause the compiler to throw errors due to the bit shifting. So I couldn't figure out how to get the following pins setup: 
9 SCLK 
10 MOSI
39 DC/A0
40 CS 
38 RESET

I did get a little confused with the page buffer, but since the Full Page did render something with the Arduino sketch I hadn't worried about that too much, though I'll implement it right away.

ToastyStoemp, 2020-09-04:
Thank you for the swift reply,

I originally did follow the u8g2as7, great stuff to get me going with atmel studio, however I couldn't really figure out the pin mapping for it.
changing the pin numbers on https://github.com/olikraus/u8g2/blob/master/sys/avr/as7/main.c#L46 and others to be anything higher than 9 cause the compiler to throw warnings, due to the bit shifting. So I wasn't sure if that would cause any problems: 
9 SCLK 
10 MOSI
39 DC/A0
40 CS 
38 RESET

in the Arduino sketch I was able to define the pins as next:
`U8G2_SH1122_256X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ A2, /* dc=*/ A3, /* reset=*/ A4);`

I did get a little confused with the page buffer, but since the Full Page did render something with the Arduino sketch I hadn't worried about that too much, though I'll implement it right away.

current main.c file: 
```
/*
 * based on code by ing_mauro
 */ 

#define F_CPU 16000000UL
#include <avr/io.h>
#include <util/delay.h>
#include "u8g2.h"
#include "u8x8.h"

// PB1=9 SCLK 
// PB2=10 MOSI
// PF4=39 DC/A0
// PF1=40 CS 
// PF5=38 RESET 

u8g2_t u8g2;

#define DISPLAY_CLK_DIR DDRB
#define DISPLAY_CLK_PORT PORTB
#define DISPLAY_CLK_PIN 9

#define DISPLAY_DATA_DIR DDRB
#define DISPLAY_DATA_PORT PORTB
#define DISPLAY_DATA_PIN 10

#define DISPLAY_CS_DIR DDRB
#define DISPLAY_CS_PORT PORTB
#define DISPLAY_CS_PIN 40

#define DISPLAY_DC_DIR DDRB
#define DISPLAY_DC_PORT PORTB
#define DISPLAY_DC_PIN 39

#define DISPLAY_RESET_DIR DDRB
#define DISPLAY_RESET_PORT PORTB
#define DISPLAY_RESET_PIN 38

#define P_CPU_NS (1000000000UL / F_CPU)

u8g2_t u8g2;

uint8_t u8x8_avr_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	uint8_t cycles;

	switch(msg)
	{
		case U8X8_MSG_DELAY_NANO:     // delay arg_int * 1 nano second
		// At 20Mhz, each cycle is 50ns, the call itself is slower.
		break;
		case U8X8_MSG_DELAY_100NANO:    // delay arg_int * 100 nano seconds
		// Approximate best case values...
		#define CALL_CYCLES 26UL
		#define CALC_CYCLES 4UL
		#define RETURN_CYCLES 4UL
		#define CYCLES_PER_LOOP 4UL

		cycles = (100UL * arg_int) / (P_CPU_NS * CYCLES_PER_LOOP);

		if(cycles > CALL_CYCLES + RETURN_CYCLES + CALC_CYCLES)
		break;

		__asm__ __volatile__ (
		"1: sbiw %0,1" "\n\t" // 2 cycles
		"brne 1b" : "=w" (cycles) : "0" (cycles) // 2 cycles
		);
		break;
		case U8X8_MSG_DELAY_10MICRO:    // delay arg_int * 10 micro seconds
		for(int i=0 ; i < arg_int ; i++)
		_delay_us(10);
		break;
		case U8X8_MSG_DELAY_MILLI:      // delay arg_int * 1 milli second
		for(int i=0 ; i < arg_int ; i++)
		_delay_ms(1);
		break;
		default:
		return 0;
	}
	return 1;
}

uint8_t u8x8_avr_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	// Re-use library for delays

	switch(msg)
	{
		case U8X8_MSG_GPIO_AND_DELAY_INIT:  // called once during init phase of u8g2/u8x8
		DISPLAY_CLK_DIR |= 1<<DISPLAY_CLK_PIN;
		DISPLAY_DATA_DIR |= 1<<DISPLAY_DATA_PIN;
		DISPLAY_CS_DIR |= 1<<DISPLAY_CS_PIN;
		DISPLAY_DC_DIR |= 1<<DISPLAY_DC_PIN;
		DISPLAY_RESET_DIR |= 1<<DISPLAY_RESET_PIN;
		break;              // can be used to setup pins
		case U8X8_MSG_GPIO_SPI_CLOCK:        // Clock pin: Output level in arg_int
		if(arg_int)
		DISPLAY_CLK_PORT |= (1<<DISPLAY_CLK_PIN);
		else
		DISPLAY_CLK_PORT &= ~(1<<DISPLAY_CLK_PIN);
		break;
		case U8X8_MSG_GPIO_SPI_DATA:        // MOSI pin: Output level in arg_int
		if(arg_int)
		DISPLAY_DATA_PORT |= (1<<DISPLAY_DATA_PIN);
		else
		DISPLAY_DATA_PORT &= ~(1<<DISPLAY_DATA_PIN);
		break;
		case U8X8_MSG_GPIO_CS:        // CS (chip select) pin: Output level in arg_int
		if(arg_int)
		DISPLAY_CS_PORT |= (1<<DISPLAY_CS_PIN);
		else
		DISPLAY_CS_PORT &= ~(1<<DISPLAY_CS_PIN);
		break;
		case U8X8_MSG_GPIO_DC:        // DC (data/cmd, A0, register select) pin: Output level in arg_int
		if(arg_int)
		DISPLAY_DC_PORT |= (1<<DISPLAY_DC_PIN);
		else
		DISPLAY_DC_PORT &= ~(1<<DISPLAY_DC_PIN);
		break;
		
		case U8X8_MSG_GPIO_RESET:     // Reset pin: Output level in arg_int
		if(arg_int)
		DISPLAY_RESET_PORT |= (1<<DISPLAY_RESET_PIN);
		else
		DISPLAY_RESET_PORT &= ~(1<<DISPLAY_RESET_PIN);
		break;
		default:
		if (u8x8_avr_delay(u8x8, msg, arg_int, arg_ptr))	// check for any delay msgs
		return 1;
		u8x8_SetGPIOResult(u8x8, 1);      // default return value
		break;
	}
	return 1;
}

int main(void) {
	u8g2_Setup_sh1122_256x64_1(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8x8_avr_gpio_and_delay );
	u8g2_InitDisplay(&u8g2);		//Send initialization code to the display
	u8g2_SetPowerSave(&u8g2, 0);
	
	u8g2_FirstPage(&u8g2);
	do {
		u8g2_SetFont(&u8g2, u8g2_font_6x10_tf);
		u8g2_DrawStr(&u8g2, 0, 0, "Test");
	} while ( u8g2_NextPage(&u8g2) );
	
	//Infinite loop didn't make a difference
	//while(1){
	//}
}
```

I appreciate you taking the time to look at my project

olikraus, 2020-09-04:
I think the term "PIN" in my main.c is misleading. It actually should be called "bit-position within the port". Obviously the value should be between 0 and 7:
``` C++
#define DISPLAY_CLK_DIR DDRB
#define DISPLAY_CLK_PORT PORTB
#define DISPLAY_CLK_PIN 9
```
is wrong and you probably mean
``` C++
#define DISPLAY_CLK_DIR DDRC
#define DISPLAY_CLK_PORT PORTC
#define DISPLAY_CLK_PIN 1
```
The other pins have to be changed in a similar way. 

Another problem is here:
``` C++
u8g2_DrawStr(&u8g2, 0, 0, "Test");
```
The reference point for any u8g2 text is the lower left edge. However the upper left pixel has the position 0,0. This means, the text would never be visible, instead use some y offset:
``` C++
u8g2_DrawStr(&u8g2, 0, 12, "Test");
```

ToastyStoemp, 2020-09-04:
Oh so we shift within the DDRB type. Right that makes sense

olikraus, 2020-09-04:
> //Edit:
Updated pin types from DDRB -> DDRF, and PORTB -> PORTF where needed without a change in result

But still the pin number denotes the bit position within the port and must be between 0 and 7

ToastyStoemp, 2020-09-04:
Sorry, but I must be doing something wrong.

I've tried a few combinations but so far to no avail.
```
// PB1=9 SCLK 
// PB2=10 MOSI
// PF1=40 CS 
// PF4=39 DC/A0
// PF5=38 RESET 

u8g2_t u8g2;

#define DISPLAY_CLK_DIR DDRB
#define DISPLAY_CLK_PORT PORTB
#define DISPLAY_CLK_PIN 1

#define DISPLAY_DATA_DIR DDRB
#define DISPLAY_DATA_PORT PORTB
#define DISPLAY_DATA_PIN 2

#define DISPLAY_CS_DIR DDRF
#define DISPLAY_CS_PORT PORTF
#define DISPLAY_CS_PIN 1

#define DISPLAY_DC_DIR DDRF
#define DISPLAY_DC_PORT PORTF
#define DISPLAY_DC_PIN 4

#define DISPLAY_RESET_DIR DDRF
#define DISPLAY_RESET_PORT PORTF
#define DISPLAY_RESET_PIN 5
```

I also tried lowering the DC and RESET pin to 2 and 3 since PF2 and PF3 aren't defined in the iom32u4.h file.

I've included a picture of my schematic, but I'm rather sure the pin mapping should be correct now 
[image]

Once again I very much appreciate the support, I've been fidling with this for a month now and I haven't gotten anywhere.

olikraus, 2020-09-04:
At least the schematic now fits to the define statements. Hope the spi subsystem of the avr is disabled.

ToastyStoemp, 2020-09-04:
Sorry I don't know what that means.
There are no other SPI devices connected to the MCU

Do you mean if the MCU is set as Master rather than slave?
```
4. Select either master or slave mode

To place the SPI module into Master mode, the MSTR bit in the SPCR register should be set to 1.

(Note - If SS is configured as an input and is driven low while MSTR is set, MSTR will be cleared, and SPIF in SPSR will become set. The user will then have to set MSTR to re-enable SPI Master mode.)
```

olikraus, 2020-09-04:
> Sorry I don't know what that means.

You have attached the SPI pins of the display to the corresponding SPI pins of your AVR. This is basically a good idea, BUT at the moment the pins are not driven by the SPI subsysten, but instead via GPIO ports.

Remember: A output pin of a controller might have multiple functions, so you could set the output level via a GPIO registor our you could activate e.g. SPI subsystem to drive the pin.

Now: If the SPI subsystem is activated, then the SPI subsystem has control over the output level, in such a case, writing to the PORT will not do anyting.
So i just wanted to say, please ensure that the SPI subsystem is inactive.

ToastyStoemp, 2020-09-04:
Did some research to fully understand what you're asking, thank you for short explanation.

 Following the datasheet of the Atmega32u4 it sounds like the SPI subsystem is by default disabled:/
[image]
source: http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7766-8-bit-AVR-ATmega16U4-32U4_Datasheet.pdf Page 180

On this page http://medesign.seas.upenn.edu/index.php/Guides/MaEvArM-gpio I found more information about setting the input and output of ports.
But even with those changes;

```
int main(void) {
	
	DDRB = (1<<DDB1)|(1<<DDB2);		//Set MOSI, SCK, CS, DC and RES as output
	DDRF = (1<<DDF1)|(1<<DDF4)|(1<<DDF5);
	
	SPCR = 0;	// Disable SPI
	
	u8g2_Setup_sh1122_256x64_1(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8x8_avr_gpio_and_delay );
	u8g2_InitDisplay(&u8g2);		//Send initialization code to the display
	u8g2_SetPowerSave(&u8g2, 0);	//Wake up display
	
	u8g2_FirstPage(&u8g2);
	do {
		u8g2_SetFont(&u8g2, u8g2_font_ncenB08_tr);
		u8g2_DrawStr(&u8g2, 0, 24, "Test");
	} while ( u8g2_NextPage(&u8g2) );
	
}
```

I couldn't get anything to display.

ToastyStoemp, 2020-09-04:
Alternatively isn't there an easier way to implement the spi communication using the spi subsystem. I understand that this method adds flexibility between the various chips and displays you're trying to support. But I personally feel like the code would be much easier to understand and implement.

olikraus, 2020-09-04:
> Alternatively isn't there an easier way to implement the spi communication using the spi subsystem.

Of course, but you need another callback function. In detail, you need to replace "u8x8_byte_4wire_sw_spi" procedure.
Details are described here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-spi-communication

Nevertheless, using the SW SPI approach with GPIO only usually is simpler to ensure that everything works correctly. Only once you have something on the screen, it might make sense to switch to HW SPI (which will be also much faster).

I personally think that there is a wiring problem in the connection between your ATMEGA and the display.

ToastyStoemp, 2020-09-04:
But the Oled does work when using the Arduino examples.
I'll look into the alternative SPI setup.

olikraus, 2020-09-04:
> But the Oled does work when using the Arduino examples.

Great, then the OLED works. But do you use the same hardware setup for your C-Code tests? I mean, you will disconnet the OLED from the Arduino and reconnect it to something else, right? Are you sure, that this other hardware works correctly?
Do the signals arrive at the display? 

- During debugging, do not assume something... prove it.
- If you believe something is right, you won't see what's wrong.
- Take baby steps
- Do not debug everything at once
- Take a break

ToastyStoemp, 2020-09-04:
I'm using the same hardware on both tests.
I'm using the same program to flash hex files.
Currently out of the country, but should be back home on Monday, could hook up a scope to verify if the pins are outputting any data.

I can't stress enough how much I appreciate your support. Currently double triple checking my PCB in software to make 100% sure I'm giving it the right pins. I agree that we're at a point where it almost can't be the code anymore. Your example code has worked for other people, so why wouldn't it work for me. I have patience, but am excited to finally make progress on a project that has been pestering me for a while.

We'll get to the bottom of this, and when we do I'll share the details of every single step so that the next person can learn from the mistakes made :)

ToastyStoemp, 2020-09-04:
I did discover a potential issue; I'm using some F ports for the CS DC and Reset.
However the datasheet states:
```
Port F also serves the functions of the JTAG interface.
...
When the JTAG
interface is enabled, this pin can not be used as an I/O pin.
```

So I checked, and it looks like Jtag is enabled by default on the Atmega32u4. I'll have to wait till Monday to change the fuse settings. But it looks like that could be the issue. How Arduino get's around this is beyond me.

olikraus, 2020-09-04:
> How Arduino get's around this is beyond me.

That's the biggest advantage of Arduino: It will take care and remove many obstacles. No need to read several 100 pages in the datasheet. Drawback is a lot code and a huge performance loss.

ToastyStoemp, 2020-09-07:
Hey, I've finally gotten the code to work on the Atmega32u4. Didn't even need to do any of the JTAG disabling or anything. Turned out it was the wiring after all, damn I feel stupid. I double triple quadruple checked the wiring on multiple occasions, but kept using the correct PCB file, though updated schematic file which included the fix for my display being upside down. My apologies for the time spend helping me. Once again thank you for the support!

closed 2020-09-07 by ToastyStoemp

olikraus, 2020-09-07:
:-D 

I would say: Of all similar questions here, 95% are wiring errors ;-)

I am happy you solved the problem.