olikraus/u8g2 #1274: U8G2 Lib with two display same adress like two wire or different SCL/SDA

Rothpe, 2020-09-04:
Hi,

i use OLED Display 0,49" 64x32 with constructor:
U8G2_SSD1306_64X32_1F_F_HW_I2C u8g2(U8G2_R0, /* reset=*/U8X8_PIN_NONE); and it works fine.
Now i will use two display with same adress.
Is it possible to update Library for this constructor:
U8G2_SSD1306_64X32_1F_F_HW_I2C u8g2(U8G2_R0, SCL1, SDA1, /* reset=*/U8X8_PIN_NONE);
U8G2_SSD1306_64X32_1F_F_HW_I2C u8g2(U8G2_R0, SCL2, SDA2, /* reset=*/U8X8_PIN_NONE);
to use different ports for these displays?

Or is it possible to use U8G2 Library with two wire configuration like Adafruit Library SSD1306?
For example:

TwoWire i2cA = TwoWire(0);
TwoWire i2cB = TwoWire(1);

Adafruit_SSD1306 displayA(SCREEN_WIDTH, SCREEN_HEIGHT, &i2cA, -1);
Adafruit_SSD1306 displayB(SCREEN_WIDTH, SCREEN_HEIGHT, &i2cB, -1);

  i2cA.begin(SDA1,SCL1,400000);
  i2cB.begin(SDA2,SCL2,400000);

  displayA.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  displayB.begin(SSD1306_SWITCHCAPVCC, 0x3C);

Thanks for your support in advance

Best regards

Peter

olikraus, 2020-09-04:
You can use SW_I2C instead of HW_I2C. With SW_I2C you can use any pins (but display update will be slower).
There is also a constructor called 2ND_I2C which will use Wire1 if available.

Rothpe, 2020-09-05:
Hi, thanks for your quick response and your great support again for me.
I changed constructo to:
U8G2_SSD1306_64X32_1F_F_HW_I2C Display1(U8G2_R0, /* reset=*/U8X8_PIN_NONE);
U8G2_SSD1306_64X32_1F_F_2ND_HW_I2C Display2(U8G2_R0, /* reset=*/U8X8_PIN_NONE);

and:
  Wire.begin(SDA1, SCL1, 400000);
  Wire1.begin(SDA2, SCL2, 400000);
but nothing display on Display 2

Can you pls help where i can configure SCL2(16) and SDA2(17) .
I use ESP32.

olikraus, 2020-09-05:
ESP 32 does not support 2nd hw.

Rothpe, 2020-09-05:
this is the reason why it does'nt work
Do you know which Arduino i can use for this funktionality?

olikraus, 2020-09-06:
Support should be there for all Arduino boards which define WIRE_INTERFACES_COUNT as 2 or higher.
There had been several discussions on this topic, like here: https://github.com/olikraus/u8g2/issues/961

I think the Due has two I2C ports.

Rothpe, 2020-09-08:
Thanks for your support.
I think for my point of view there is no possibility to use 2 OLED 0,49 - 64x32 with U8G2 Lib.
Also it is not necessary to use Adafruit SSD1306 Lib.
I have to check an other way.

closed 2020-09-08 by Rothpe