olikraus/u8g2 #1276: U8G2_T6963_240X128_2_8080 double graphic

t3moses, 2020-09-07:
I am using an Arduino UNO R3 running at 16MHz with a Newhaven NHD-240128WG-ATFH-VZ# display.  It is a 240x128 LCD based on the RAIO RA6963L2NA controller.

Here's my code:
````
#include <U8g2lib.h>

#define ROTATION U8G2_R0 // No rotation, landscape.
#define D0 8
#define D1 9
#define D2 2
#define D3 3
#define D4 4
#define D5 5
#define D6 6
#define D7 7
#define _RD 14 // Must be set HIGH.
#define _WR 15 // AKA ENABLE
#define _DC 16 // AKA CD (Command or data)
#define _CS 17 // AKA CE (chip enable)
#define _RESET 18 // AKA RST

#define _BL 10 // Backlight, LOW = ON.

U8G2_T6963_240X128_2_8080 display(ROTATION, D0, D1, D2, D3, D4, D5, D6, D7, _WR, _CS, _DC, _RESET); // 480 byte page buffer

void setup(void) {

  pinMode(_RD, OUTPUT);
  digitalWrite(_RD, HIGH);

  pinMode(_BL, OUTPUT);
  digitalWrite(_BL, LOW);
  
  display.begin();
  
}

void loop(void) {

  display.clearBuffer();
  
  display.firstPage();
  do {

    for ( uint8_t i = 0; i < 12; i++ ) {

      display.drawFrame(2, 14 + i * 8, i, 8);

    }
    
  } while ( display.nextPage() );
  
  halt();

}

void halt() {

  delay(5000);
  digitalWrite(_BL, HIGH);
  while(true);
  
}
````
You can see the result in the attached photo.

The correct graphic appears on the left of the screen and an echo appears at x=180 (i.e. 240 x 6/8).  The top line of the graphic is missing from the echo.

It appears that the display is discarding bits 6 and 7 from each data byte.

Has anyone seen this before?  Is it possible that the display input register is being overwritten before it has processed bits 6 and 7 from the previous register contents?  Should I contact support at Newhaven?

Grateful for any help.  Best wishes.  Tim.

[image]

t3moses, 2020-09-07:
Slight correction ... the photo was created with a different version of the code; one that read and displayed the display width.  Nevertheless, I think the issue is clear.

olikraus, 2020-09-07:
You must set FS to GND.

t3moses, 2020-09-07:
First of all, thank you very much for your quick response.

I was about to reply that I had already tried FS set both to 0v and to 5v.  But, I tried setting it back to 0v, and Hallelujah!  It is now working.

Thanks again.  Best wishes.  Tim.

closed 2020-09-08 by olikraus