olikraus/u8g2 #1277: character shifts when using updateDisplayArea with SH1122

Jack-XHP, 2020-09-07:
Hello, I am using an i2c OLED display with SH1122. The display is running in full frame mode and is working perfectly when using sendBuff. However, when I using updateDisplayArea, each character will be separated into many parts, and each part will be shifted by certain pixels. Here is my code, I use the same function on a SSD1306 oled display and both sendBuff and update DisplayArea both works correctly. So I am not sure where went wrong for SH1122.

`#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>
U8G2_SH1122_256X64_F_HW_I2C u8g3(U8G2_R0);
void TCA9548A(uint8_t bus)
{
  Wire.beginTransmission(0x70);  // TCA9548A address is 0x70
  Wire.write(1 << bus);          // send byte to select bus
  Wire.endTransmission();
}
void disNumBig(int address, char* text){
  TCA9548A(address);
  u8g3.clearBuffer();          // clear the internal memory 
  u8g3.setCursor(0, 30);
  u8g3.print(text);
  u8g3.sendBuffer(); 
}
void disScrtachStr1(int address, char* text){
  TCA9548A(address);
  u8g3.setCursor(0, 30);
  u8g3.print(text);
  u8g3.updateDisplayArea(0,0,32,8);
}
void setup() 
{
  
  Wire.begin();
  
  TCA9548A(5);
  u8g3.begin();
  u8g3.clearBuffer(); 
  u8g3.sendBuffer(); 
  u8g3.setFont(u8g2_font_inb30_mf);
}
void loop()
{ 
  
  disScrtachStr1(5,"A");
  disNumBig(5,"A");
}

`

olikraus, 2020-09-07:
Difficult... ok, the clearBuffer() might be missing in disScrtachStr1().
Can you sent a picture of the output for  disScrtachStr1 (with the clearBuffer() as first command)

Jack-XHP, 2020-09-07:
I  kind of find a way to fix the issue partially. I found the shift is correlated with the width of display area, and with the correct width, I can print a readable text, but not every length will work in this method.
u8g3.updateDisplayArea(0,0,8,4);
[image]
u8g3.updateDisplayArea(0,0,16,4);
[image]

olikraus, 2020-09-07:
Difficult. Now there is a picture, but i guess the code is different. Moreover it looks like that only every second line is printed.
Is this normal? I mean, if only every second line is printed, then this should be fixed first.

theeprawn, 2020-09-26:
Hi, I had issue with UpdateDisplayArea with SSH1122 and SSD1362.  I cured it by adding a horizontal buffer flag in u8g2_GetU8x8(u8g2)->display_info->pixel_byte_horizontal.
The issue is u8x8 uses bytes vertically, SSH1122 and SSD1362 have horizontal bytes, which u8g2 switches, but UpdateDisplayArea  doesnt. I couldnt see a way to see which way buffer switched so just added the flag so can switch between buffer layout in UpdateDisplayArea.

theeprawn, 2020-09-26:
void u8g2_UpdateDisplayArea(u8g2_t *u8g2, uint8_t  tx, uint8_t ty, uint8_t tw, uint8_t th)
{
  uint16_t page_size;
  uint8_t *ptr;
  uint8_t *tptr;
  /* check, whether we are in full buffer mode */
  if ( u8g2->tile_buf_height != u8g2_GetU8x8(u8g2)->display_info->tile_height )
    return; /* not in full buffer mode, do nothing */

  page_size = u8g2->pixel_buf_width;  /* 8*u8g2_GetU8x8(u8g2)->display_info->tile_width */
    
  ptr = u8g2_GetBufferPtr(u8g2);
  if(u8g2_GetU8x8(u8g2)->display_info->pixel_byte_horizontal==0){
    ptr += tx*8; /* 8 bytes across per tile, stacked vertically */
  }else{
    ptr += tx; /* 1 byte across per tile, stacked horizontally */
  }
  ptr += page_size*ty;
  
  while( th > 0 )
  {
    u8x8_DrawTile( u8g2_GetU8x8(u8g2), tx, ty, tw, ptr );
    ptr += page_size;
    ty++;
    th--;
  }
}

theeprawn, 2020-09-26:
I rewritten display code as well. but want ask something about how can display driver code know if u8g2 is using it, as has same issue(u8x8 uses verticaly bytes). 
I tried using #ifndef U8G2_H, but it is never defined even though u8g2 sets it. I not sure why as first thing set in u8g2.h

or would be better I just pull request with know issues?

olikraus, 2020-10-01:
It is true, that SH1122 does NOT support updateDisplayArea die to the different low level byte output strategy. I added a note to https://github.com/olikraus/u8g2/wiki/u8g2reference#updatedisplayarea

Whether horizontal or vertical mode is used, is defined in the setup procedure, which is in turn generated: 

https://github.com/olikraus/u8g2/blob/cd1027c84e3f683b9c8bf507cb864e5dd2702c98/tools/codebuild/codebuild.c#L294-L308

theeprawn, 2020-10-01:
Yeah, I saw its done in setup in u8g2 but I couldnt find away to see which function is used. I tried
if( u8g2->ll_hvline == u8g2_ll_hvline_vertical_top_lsb ){
but it always came back false, same with 
if(u8g2->ll_hvline == &u8g2_ll_hvline_horizontal_right_lsb){
So I added the extra pixel_byte_horizontal to display_info. I sure there a way to compare but I cant debug inside u8g2 or u8x8 in Ardunio framework log_d() and Serial.prinf() arent avaliable in u8g2 or u8x8.
Hope I not messed up the Pull requests, added glyph scroll/rotate, then put changes to displayupdate area and sh1122 and ssd1362, but it put them all in same pull request, sorry I not use to Github, just wanted to share some code to help.

theeprawn, 2020-10-02:
Ah, ignore me about the below not working, my bad was testing on product with updatedisplayarea starting at left edge(tx=0)
if( u8g2->ll_hvline == u8g2_ll_hvline_vertical_top_lsb ){
So now dont need pixel_byte_horizontal info.
Works fine in updated SH1122 driver and SSD1362. I tested both these with SPI wire and I2C, recommend SPI 3-wire, screen refresh rate slow with I2C (about 3full fps) because on greyscale on 2 pixels per byte. With SPI you get over 30full fps at 10MHz clk, even faster with 40MHz, but that outside displays IC's specification.

void u8g2_UpdateDisplayArea(u8g2_t *u8g2, uint8_t  tx, uint8_t ty, uint8_t tw, uint8_t th)
{
  uint16_t page_size;
  uint8_t *ptr;

  /* check, whether we are in full buffer mode */
  if ( u8g2->tile_buf_height != u8g2_GetU8x8(u8g2)->display_info->tile_height )
    return; /* not in full buffer mode, do nothing */

  page_size = u8g2->pixel_buf_width;  /* 8*u8g2_GetU8x8(u8g2)->display_info->tile_width */
    
  ptr = u8g2_GetBufferPtr(u8g2);

  if(u8g2->ll_hvline == u8g2_ll_hvline_vertical_top_lsb){
    ptr += tx * 8; /* 8 bytes across per tile, stacked vertically */
  }else{
    ptr += tx; /* 1 byte across per tile, stacked horizontally */
  }
  ptr += page_size*ty;

  while( th > 0 )
  {
    u8x8_DrawTile( u8g2_GetU8x8(u8g2), tx, ty, tw, ptr );
    ptr += page_size;
    ty++;
    th--;
  }
}

olikraus, 2020-10-03:
Just a quick feedback. 
Thanks a lot for you in depth coding and updates. 
However I am not sure whether i can integrate the pull requires. A pull request just for the display controller update would be great, but maybe I can add this manually. On the other side I have some doubts with the character shift.
I probably need to investigate this, when I have a little bit more time.

theeprawn, 2020-10-09:
I try and write a demo ardunio sketch for scrolling abd try it on a ssd1306. Think your like it, I was surprise at effects. I only wrote to animate a status icon, seemed more approperiate to scroll in driver than in application, but effect on other fonts look interesting/useful so extended with x and gap.
I will see if I can sort the u8x8 vs u8g2 driver detection out, then ssd1362 and sh1122 will be done to auto detect which way tile bits are. Then see if I can pull request, will I need to make another branch to make a new pull request, I tried a pull request after the glyph scrolling, but it put the drivers in with the scrolling changes.

olikraus, 2022-03-04:
I did some investigation on this problem today. 
Unfortunately this the SH1122 controller is not compatible with the updateDisplayArea command. There is nothing what I can do here (except rewriting larger parts of u8g2). Closing this...

closed 2022-03-04 by olikraus