olikraus/u8g2 #1278: SPI DMA implementation

margce, 2020-09-08:
Hi,
Not an issue just looking for more info in what would the best way to implement DMA transmission on my port.
I’m currently using a LCD with ST75256 (256x160) driver on 4w spi mode on a STM32F3, everything works fine but would like to add DMA to free some resources and optimise the port.

Thanks,
Martin

olikraus, 2020-09-08:
I assume that this is a none-Arduino environment.

I am not sure how you could save resources with DMA. 
Concerning RAM consumption, you should use page mode instead of full buffer mode.
Regarding speed, the first thing should be the implementation of hardware SPI transfer. Details are here:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-spi-communication

Unless you use a kind of a double buffer mechanism (which will increase RAM consumption) DMA will not increase speed or reduce code size: You still need to wait until the complete U8g2 buffer is transfered otherwise you run into the risk, that u8g2 overwrites data, which are not yet transfered. As a result, there is no difference between normal SPI transfer and DMA supported SPI transfer. In both cases you need to wait until everything is transfered.

margce, 2020-09-08:
That's correct, I'm running on a custom board and already running on hardware SPI. I'm not concerned about RAM consumption but rather trying to speed up the process of drawing all the tiles.
I understand that I'll need to wait for the DMA transfer to finish but I'm sure there should a speed increase by sending 32 bytes in a single DMA transfer than sending 1 at a time.

From uint8_t u8x8_d_st75256_jlx256160(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) DRAW_TILE

`do {
			c = ((u8x8_tile_t *) arg_ptr)->cnt;
			ptr = ((u8x8_tile_t *) arg_ptr)->tile_ptr;
			/* SendData can not handle more than 255 bytes, treat c > 31 correctly  */
			if (c > 31) {
				u8x8_cad_SendData(u8x8, 248, ptr); /* 31*8=248 */
				ptr += 248;
				c -= 31;
			}

			u8x8_cad_SendData(u8x8, c * 8, ptr);
			arg_int--;
		} while (arg_int > 0);`

I think I can optimise that code to set the DC pin and then perform a DMA transfer, couldn't I? I'll try it and measure the speed as well and report back.

margce, 2020-09-09:
I just made a quick test,

1. Using the standard 4wire SPI hw port it takes about ~16ms to send the entire buffer to the screen. 
2. Updating U8X8_MSG_DISPLAY_DRAW_TILE to use a SPI DMA transfer and optimising calls to set/reset CD pin it reduced the transfer time to under 5ms!

I understand that these modifications are very specific to my port but i think it's worth it in my case.

Thanks!
Martin

olikraus, 2020-09-09:
Well, sounds good.

closed 2020-09-10 by margce

imval, 2020-11-26:
@margce would you mind sharing your code ?

margce, 2020-11-27:
> @margce would you mind sharing your code ?

Yes, but please consider that that my implementation is specific for my platform (ST75256 LCD 256x160px, custom STM32F3 board), I've also modified the lower u8x8 routines in favour of more speed. Before I said that the screen refreshes in 5ms but it much less than that. 
You might also consider some safe guards in some of the functions to prevent the code getting stuck in some of those while(); i just haven't updated those with some timeouts.

In any case here is how my pin initialisation looks like, notice that DMA_InitStructure is a global variable, I set the fixed config during the setup and and then update the DMA information before starting the transfer. 
```
void u8x8_gpio_init( void ){
    GPIO_InitTypeDef GPIO_InitStructure;
    SPI_InitTypeDef SPI_InitStructure;

    RCC_APB2PeriphResetCmd(LCD_SPI_RCC, ENABLE);
    RCC_APB2PeriphResetCmd(LCD_SPI_RCC, DISABLE);
    RCC_APB2PeriphClockCmd(LCD_SPI_RCC, ENABLE );

    //SCK & MOSI
    GPIO_InitStructure.GPIO_Pin = LCD_SCL_PIN | LCD_MOSI_PIN;
    GPIO_InitStructure.GPIO_Speed = LCD_SCL_SPEED;
    GPIO_InitStructure.GPIO_Mode = LCD_SCL_MODE;
    GPIO_InitStructure.GPIO_OType = LCD_GPIO_OTYPE;
    GPIO_InitStructure.GPIO_PuPd = LCD_GPIO_PUPD;
    GPIO_Init(LCD_GPIO, &GPIO_InitStructure);
    GPIO_PinAFConfig( LCD_GPIO, LCD_SCL_PIN_SOURCE, LCD_SCL_AF );
    GPIO_PinAFConfig( LCD_GPIO, LCD_MOSI_PIN_SOURCE, LCD_MOSI_AF );
    //CD, NSS, RST
    GPIO_InitStructure.GPIO_Pin = LCD_CD_PIN | LCD_NSS_PIN | LCD_RESET_PIN;
    GPIO_InitStructure.GPIO_Mode = LCD_CD_MODE;
    GPIO_Init(LCD_GPIO, &GPIO_InitStructure);

    //SPI config
	SPI_InitStructure.SPI_Direction = SPI_Direction_1Line_Tx;
	SPI_InitStructure.SPI_Mode = SPI_Mode_Master;
	SPI_InitStructure.SPI_DataSize = SPI_DataSize_8b;
	SPI_InitStructure.SPI_CPOL = SPI_CPOL_Low;
	SPI_InitStructure.SPI_CPHA = SPI_CPHA_1Edge;
	SPI_InitStructure.SPI_NSS = SPI_NSS_Soft;
	SPI_InitStructure.SPI_BaudRatePrescaler = SPI_BaudRatePrescaler_2;
	SPI_InitStructure.SPI_FirstBit = SPI_FirstBit_MSB;
	SPI_InitStructure.SPI_CRCPolynomial = 7;
	SPI_Init( LCD_SPI, &SPI_InitStructure);

	/*!< Enable the SPI  */
	SPI_Cmd( LCD_SPI, ENABLE );

	/* DMA channel Tx of SPI Configuration */
	DMA_DeInit(LCD_SPI_TX_DMA_CHANNEL);
	DMA_InitStructure.DMA_PeripheralBaseAddr = (uint32_t)&LCD_SPI_DR;
	DMA_InitStructure.DMA_DIR = DMA_DIR_PeripheralDST;
	DMA_InitStructure.DMA_PeripheralInc = DMA_PeripheralInc_Disable;
	DMA_InitStructure.DMA_MemoryInc = DMA_MemoryInc_Enable;
	DMA_InitStructure.DMA_PeripheralDataSize = DMA_PeripheralDataSize_Byte;
	DMA_InitStructure.DMA_MemoryDataSize = DMA_MemoryDataSize_Byte;
	DMA_InitStructure.DMA_Mode = DMA_Mode_Normal;
	DMA_InitStructure.DMA_Priority = DMA_Priority_Medium;
	DMA_InitStructure.DMA_M2M = DMA_M2M_Disable;
}
```

Here is how I send a single byte over SPI

```
void u8x8_spi_byte_send( uint8_t byte )
{
	/*!< Loop while DR register in not empty */
	while((SPI_I2S_GetFlagStatus(LCD_SPI, SPI_I2S_FLAG_TXE) == RESET));
	/*!< Send byte through the SPI peripheral */
	SPI_SendData8(LCD_SPI, byte );
}

```
[... 84 lines cut ...]
```
uint8_t u8x8_cad_011(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	switch (msg) {
	case U8X8_MSG_CAD_SEND_CMD:
		lcd_cd_rst();
		u8x8_spi_byte_send( arg_int );
		break;
	case U8X8_MSG_CAD_SEND_ARG:
		lcd_cd_set();
		u8x8_spi_byte_send( arg_int );
		break;
	case U8X8_MSG_CAD_SEND_DATA:
		lcd_cd_set();
		//u8x8_byte_SendBytes(u8x8, arg_int, arg_ptr);
		//break;
		/* fall through */
	case U8X8_MSG_CAD_INIT:
	case U8X8_MSG_CAD_START_TRANSFER:
	case U8X8_MSG_CAD_END_TRANSFER:
		return u8x8->byte_cb(u8x8, msg, arg_int, arg_ptr);
	default:
		return 0;
	}
```

imval, 2020-11-27:
Wow thank you so much, I will try to implement it for SSD1306!

michaesc, 2021-07-27:
Were you @imval able to implement the DMA strategy from @margce for SSD1306 displays?

My goal is similar: After having hardware DMA working correctly on STM32, to improve it by using DMA and double buffers.

Unfortunately the information in this bug report is quite complicated.

amso-pjco, 2024-11-19:
I am using an LCD with the UC1701 driver and the function u8x8_d_uc1701_mini12864. I want to optimize my code to make it faster by using DMA. My current code is as follows:

```c
#include "main.h"
/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
/* LCD -----------------------------------------------------------------------*/    
#include "u8g2.h"
/* USER CODE END Includes */

/* Private define ------------------------------------------------------------*/
/* USER CODE BEGIN PD */
#define TX_TIMEOUT	100
/* USER CODE END PD */
/* Private variables ---------------------------------------------------------*/

SPI_HandleTypeDef hspi3;
DMA_HandleTypeDef hdma_spi3_tx

/* Private function prototypes -----------------------------------------------*/
void SystemClock_Config(void);
static void MX_GPIO_Init(void);
static void MX_DMA_Init(void);
static void MX_SPI3_Init(void);
/* Private user code ---------------------------------------------------------*/
/* USER CODE BEGIN 0 */
u8g2_t u8g2;

uint8_t u8x8_stm32_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	/* STM32 supports HW SPI, Remove unused cases like U8X8_MSG_DELAY_XXX & U8X8_MSG_GPIO_XXX */
	switch(msg)
	{
	case U8X8_MSG_GPIO_AND_DELAY_INIT:
		/* Insert codes for initialization */
		break;
	case U8X8_MSG_DELAY_NANO:
		break;
	case U8X8_MSG_DELAY_100NANO:
		__NOP();
		break;
	case U8X8_MSG_DELAY_10MICRO:
		for (uint16_t n = 0; n <320; n++)
		{
			__NOP();
		}
		break;
	case U8X8_MSG_DELAY_MILLI:
		/* ms Delay */
		HAL_Delay(arg_int);
		break;
	case U8X8_MSG_GPIO_SPI_CLOCK:
		HAL_GPIO_WritePin(LCD_SCK_GPIO_Port, LCD_SCK_Pin, arg_int ? GPIO_PIN_SET : GPIO_PIN_RESET);
		break;          
	case U8X8_MSG_GPIO_SPI_DATA:
		HAL_GPIO_WritePin(LCD_MOSI_GPIO_Port, LCD_MOSI_Pin, arg_int ? GPIO_PIN_SET : GPIO_PIN_RESET);
		break;
	case U8X8_MSG_GPIO_DC:
		HAL_GPIO_WritePin(LCD_DC_GPIO_Port, LCD_DC_Pin, arg_int ? GPIO_PIN_SET : GPIO_PIN_RESET);
		break;
	case U8X8_MSG_GPIO_RESET: 
		HAL_GPIO_WritePin(LCD_RST_GPIO_Port, LCD_RST_Pin, arg_int ? GPIO_PIN_SET : GPIO_PIN_RESET);
		break;
                
	}
	return 1;
}

uint8_t u8x8_byte_stm32_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	switch(msg) {
	case U8X8_MSG_BYTE_SEND:
		// Insert codes to transmit data 
		if(HAL_SPI_Transmit(&hspi3, arg_ptr, arg_int, TX_TIMEOUT) != HAL_OK) return 0;
		break;
	case U8X8_MSG_BYTE_INIT:
		//Insert codes to begin SPI transmission 
		break;
	case U8X8_MSG_BYTE_SET_DC:
		//Control DC pin, U8X8_MSG_GPIO_DC will be called 
		u8x8_gpio_SetDC(u8x8, arg_int);
		break;
	case U8X8_MSG_BYTE_START_TRANSFER:
		// Select slave, U8X8_MSG_GPIO_CS will be called 
		u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
		HAL_Delay(1);
		break;
	case U8X8_MSG_BYTE_END_TRANSFER:
		HAL_Delay(1);
		// Insert codes to end SPI transmission 
		u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
		break;
	default:
		return 0;
	}
	return 1;
}

/* USER CODE END 0 */

int main(void)
{
  /* USER CODE BEGIN 1 */

  /* USER CODE END 1 */

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* USER CODE BEGIN Init */

  /* USER CODE END Init */

  /* Configure the system clock */
  SystemClock_Config();

  /* USER CODE BEGIN SysInit */

  /* USER CODE END SysInit */

  /* Initialize all configured peripherals */
  MX_GPIO_Init();
  MX_DMA_Init();
  MX_SPI3_Init();
  /* USER CODE BEGIN 2 */  
  u8g2_Setup_uc1701_mini12864_f(&u8g2, U8G2_R2, u8x8_byte_stm32_hw_spi, u8x8_stm32_gpio_and_delay);
  u8g2_InitDisplay(&u8g2);
  u8g2_SetPowerSave(&u8g2, 0);
  u8g2_SetContrast(&u8g2,0);
  u8g2_ClearBuffer(&u8g2);// Clear the display buffer
  u8g2_SetDrawColor(&u8g2,1);
  char buffer[20]; // For holding the display string 
  /* USER CODE END 2 */

  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
    HAL_GPIO_WritePin(GPIOA,GPIO_PIN_9,GPIO_PIN_RESET);
    i= i+0.01;
    u8g2_ClearBuffer(&u8g2);// Clear the display buffer---->[run time in loop = 2.8ms]
    u8g2_SetFont(&u8g2,u8g2_font_logisoso16_tf);// Set the font and draw the formatted string on the display
    sprintf(buffer, "%2.3f", i);// Format the number as a string---->[run time in loop = 1ms]
    u8g2_DrawStr(&u8g2, 7, 18, buffer);//---->[run time in loop = 9.4ms]
    u8g2_SendBuffer(&u8g2);// Send the buffer to the display---->[run time in loop = 56ms]
    HAL_GPIO_WritePin(GPIOA,GPIO_PIN_9,GPIO_PIN_SET);
    /* USER CODE END WHILE */
  }
  /* USER CODE END 3 */
}

I have also commented the execution time of each function next to it in the code.