olikraus/u8g2 #1282: 128x64 SSD1309 i2c problems - not wiring related

joelhauxwell, 2020-09-11:
Hello,

I have an SSD1309 graphic OLED display module hooked up to an Arduino Nano.

For hours and hours yesterday i tried to get it to work with no success and concluded that that the display was defective.

So last night I found another manufacturer of the display (or at least they look identical) and the product page features a simple library and sketch to test it.  I uploaded it and bingo, the display came to life (once id connected the reset line to pin 8 - everything else was wired up OK).

The page with the working code is here: https://www.buydisplay.com/spi-i2c-yellow-2-4-inch-arduino-raspberry-pi-oled-display-module-128x64

I have tried u8g2 with every constructor in the list for a 128x64 SSD1309 with reset and hardware i2c and none of them work.

Is there anything i can try next? I would really like to use this library.

Thanks,

Joel.

olikraus, 2020-09-13:
If you have a working lib, then also u8g2 should work. Did you update the constructor args with the correct pin numbers?

joelhauxwell, 2020-09-13:
The constructor args I have tried are as follows :

U8G2_SSD1309_128X64_NONAME2_1_HW_I2C u8g2(U8G2_R0, /* clock=*/ A5, /* data=*/ A4, /* reset=*/ 8);
screen blank

U8G2_SSD1309_128X64_NONAME0_1_HW_I2C u8g2(U8G2_R0, /* clock=*/ A5, /* data=*/ A4, /* reset=*/ 8);
screen blank

This is using the GraphicsTest.ino sketch from examples

Have i got the formatting of the args correct?

Thanks,

Joel.

olikraus, 2020-09-13:
You don't need to provide the pin number for her iwc on the Arduino nano. Your display is connected to these pins, right?

joelhauxwell, 2020-09-13:
Ive had a small amount of progress...

Using this constructor

U8G2_SSD1309_128X64_NONAME0_1_HW_I2C u8g2(U8G2_R0, /* clock=/ A5, / data=/ A4, / reset=*/ 8);

I get text on the display.  In the top left hand corner it is scrolling

drawBox
drawDisc

etc... but it doesn't draw any shapes.... see vid attached (as zip because I couldn't attach .mp4)
[display video.zip](https://github.com/olikraus/u8g2/files/5214075/display.video.zip)

At least there is something on the screen now :)

Joel.

olikraus, 2020-09-13:
Do the u8g2 examples work? Graphics test for example? I use my Mobile at the moment, I can't watch the video.

joelhauxwell, 2020-09-13:
So, it looks like only the top part of the display is working with u8g2.

If i upload the U8g2Logo sketch, i get this on the display...

[image]

But when i upload the sketch from the manufacturer, the display is ok.

[image]

Any ideas what i might be doing wrong with the u8g2 examples?

Thanks again,

Joel.

olikraus, 2020-09-13:
Looks like a page buffer vs full buffer problem. The `_1_` constructor requires picture loop/page buffer examples.

olikraus, 2020-09-13:
Or instead replace 1 with f in your constructor

joelhauxwell, 2020-09-13:
"Or instead replace 1 with f in your constructor" - That has solved it - thanks!

I don't fully understand why it has - i thought that using 1 or 2 would have been OK for the 328p as its one of the smaller microcontrollers.

Thanks for your time with this.

Joel.

olikraus, 2020-09-13:
1/2 also require different commands, that's why you only see one line.

closed 2022-03-04 by olikraus