olikraus/u8g2 #1285: My 12864 has no DC pin，how should I configure this function -- u8x8_byte_4wire_hw_spi

Charley175, 2020-09-18:
1 VDD 
2 VSS 
3 SID 
4 CS
5 CLK 
6 EX-IN2
7 NC 
8 EX-IN3 
9 EX-IN1 
10 BELL

olikraus, 2020-09-19:
I am sure the display datasheet gives sufficient advice.

Charley175, 2020-09-19:
HI！
I'm really glad you can return.
My LCD screen does not have a DC pin, what should I do?

uint8_t Data = 0xFA;
uint8_t Cmd = 0xF8;
uint8_t Clean = 0x01;
case U8X8_MSG_GPIO_DC:
if (arg_int){
HAL_SPI_Transmit(&Spi2, (uint8_t *) &Data, 1, 10000);
}else{
HAL_SPI_Transmit(&Spi2, (uint8_t *) &Cmd, 1, 10000);}
break;

Then I wanted to use the above command to tell my LCD12864, whether I want to write data or commands, but my LCD display is messed up, please help me，thanks.

Charley175, 2020-09-19:
I read the data sheet, but once I write the command, the display will be wrong, so I want some help.

Charley175, 2020-09-19:
The controller of my 12864 display is ST7920. @olikraus

olikraus, 2020-09-19:
> My LCD screen does not have a DC pin, what should I do?

Nothing, the ST7920 in SPI mode does not require the DC pin. In fact you do not need to name the DC pin on the constructor:

https://github.com/olikraus/u8g2/wiki/gallery#25-jun-2016-st7920-128x64-lcd

and

https://github.com/olikraus/u8g2/wiki/setup_tutorial#identify-digital-lines

Charley175, 2020-09-19:
@olikraus  [https://github.com/Charley175/no/blob/master/%E5%8D%8E%E6%98%BE%E6%99%B6SP-HSM12864M-16%E8%A7%84%E6%A0%BC%E4%B9%A6.pdf](url) ,This is my datasheet.

olikraus, 2020-09-19:
Another problem could be this: The display is a 5V display, however the STM32 is 3.3V device. The output voltage of the GPIO pins might not be sufficient for the ST7920.

Charley175, 2020-09-19:
@olikraus  If you read my datasheet, you will find that there is no reset pin, the wiring method is EXP3. I have considered the power supply problem. It can be displayed normally, but its display mode is too limited, so I chose to use your U8GLIB @olikraus

olikraus, 2020-09-19:
I can not read the datasheet, the link does not work for me.

> I have considered the power supply problem.

This is not sufficient. You need to think about level shifting for the GPIO pins.

> It can be displayed normally, but its display mode is too limited, so I chose to use your U8GLIB

So U8GLIB will work?

Charley175, 2020-09-19:
@olikraus My 12864 only has three wires, chip select, clock, and data, so I can’t make it work yet. My monitor has an additional 5V power supply.

olikraus, 2020-09-19:
> My 12864 only has three wires, chip select, clock, and data,

Fine, but this does not seem to be a software problem, right? So how can I help here? I can't...
If the display does not contain a reset input, then it probably has a kind of an internal reset circuit, so reset line is not required. You can skip the same. But without datasheet, it is impossible to say this.

> My monitor has an additional 5V power supply.

Great, but this is (third time that i mention this) NOT the problem. The problem is the output of the SPI & Clock data lines, which will be 3.3V without additional hardware.

> My datasheet is a PDF file, you should be able to see it when you download it.

Clicking on the link above leads to this url: https://github.com/olikraus/u8g2/issues/url
Copying the displayed link name i get a github error

closed 2022-03-04 by olikraus