olikraus/u8g2 #1286: Only half display working

TheRealArtonn, 2020-09-18:
OLED I am using. https://www.aliexpress.com/item/4000465625534.html?spm=a2g0s.9042311.0.0.f1fc4c4dIACjuD

So everything shows fine on the OLED, animations works and I am using the graphicsTest that comes with the library, but for some reason it only shows something on half of the screen.

https://media.discordapp.net/attachments/451155251580633089/756602624035389490/image0.jpg?width=508&height=677

olikraus, 2020-09-19:
Which constructor did you use?
Maybe your display is not yet supported.

Please locate file u8x8_d_sh1107.c on your harddisk. Modify the following values:
https://github.com/olikraus/u8g2/blob/5e8960d931d41a2d3c38752f44d56839039a0402/csrc/u8x8_d_sh1107.c#L208-L209

Try values like 16 or 32.

TheRealArtonn, 2020-09-19:
I was using the constructor 
//U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

So something like this ? `static const u8x8_display_info_t u8x8_sh1107_64x128_noname_display_info =
{
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 20,
  /* pre_chip_disable_wait_ns = */ 10,
  /* reset_pulse_width_ms = */ 100, 	/* sh1107: 3 us */
  /* post_reset_wait_ms = */ 100, /* far east OLEDs need much longer setup time */
  /* sda_setup_time_ns = */ 50,		/* sh1107: 15ns, but cycle time is 100ns, so use 100/2 */
  /* sck_pulse_width_ns = */ 50,	/* sh1107: 20ns, but cycle time is 100ns, so use 100/2, AVR: below 70: 8 MHz, >= 70 --> 4MHz clock */
  /* sck_clock_hz = */ 8000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 40,
  /* write_pulse_width_ns = */ 150,	/* sh1107: cycle time is 300ns, so use 300/2 = 150 */
  /* tile_width = */ 8,
  /* tile_height = */ 16,
  /* default_x_offset = */ 16,
  /* flipmode_x_offset = */ 16,
  /* pixel_width = */ 64,
  /* pixel_height = */ 128
};`

olikraus, 2020-09-19:
Yes, exactly. Will this change the display?

TheRealArtonn, 2020-09-19:
Sadly it appears the same to me. :/

olikraus, 2020-09-19:
Didn't you use the sh1107 constructor?

TheRealArtonn, 2020-09-19:
I am currently using the graphicsTest, with this outcommented: (U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);).

I went in and edited the default_x_offset and flipmode_x_offset on the (static const u8x8_display_info_t u8x8_sh1107_64x128_noname_display_info), I then saved with notepad++ and reuploaded the sketch to my arduino. Sadly nothing changed.

TheRealArtonn, 2020-09-19:
I also edited u8x8_d_sh1107.

olikraus, 2020-09-19:
Modifing the values should lead to a visual change. If not then you either modified the wrong file or Arduino IDE didn't recompile everything. Arduino IDE is a little bit special. It sometimes installs multiple instances of the same lib in different folders. It also does not always recompile everything. Sometimes you need to closed and restart Arduino IDE.

TheRealArtonn, 2020-09-19:
So I am not very good at this, but what I can gather is I am "U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);" That is SH1106, so it would not make sense or me to make changes in something named u8x8_d_sh1107.c ? "sh1107"

TheRealArtonn, 2020-09-19:
So ended up using U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); and these settings

```
static const u8x8_display_info_t u8x8_ssd1306_128x64_noname_display_info =
{
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 20,
  /* pre_chip_disable_wait_ns = */ 10,
  /* reset_pulse_width_ms = */ 100, 	/* SSD1306: 3 us */
  /* post_reset_wait_ms = */ 100, /* far east OLEDs need much longer setup time */
  /* sda_setup_time_ns = */ 50,		/* SSD1306: 15ns, but cycle time is 100ns, so use 100/2 */
  /* sck_pulse_width_ns = */ 50,	/* SSD1306: 20ns, but cycle time is 100ns, so use 100/2, AVR: below 70: 8 MHz, >= 70 --> 4MHz clock */
  /* sck_clock_hz = */ 8000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 40,
  /* write_pulse_width_ns = */ 150,	/* SSD1306: cycle time is 300ns, so use 300/2 = 150 */
  /* tile_width = */ 16,
  /* tile_hight = */ 8,
  /* default_x_offset = */ 32,
  /* flipmode_x_offset = */ 32,
  /* pixel_width = */ 128,
  /* pixel_height = */ 64
};
```
These settings pushes the image out of the screen.

TheRealArtonn, 2020-09-19:
So been playing with some numbers and the graphics element just get pushed to the left out of the screen and does not fill the missing screen up.

olikraus, 2020-09-19:
I am a little bit confused, The aliexpress link in your first post shows a SH1107 OLED.
So i have assumed that you will use the SH1107 constructor. In fact I am very sure that nither the SSD1306 nor the SH1106 will work. In fact SSD1306 or SH1106 differ a lot from the SH1107. And yes, of course if you use a SSD1306 you need to modify values for the SSD1306. Anyhow, the SSD1306 would never show an artefact shown in your picture. Truly you have an SSH1107 and you need to use the constructor for this. By using the SSD1306 constructor you can not solve the problem.

TheRealArtonn, 2020-09-19:
I might be confusing myself even more, In the title they are saying SSD1107 and just saw they were saying SH1107 further down in the description. The closest component I could find was "U8G2_SH1107_64X128_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);" But this OLEd only have SCL SDA ports, so I think this component would not work?

olikraus, 2020-09-19:
Obviously U8G2_SH1107_64X128_1_HW_I2C already exists.

TheRealArtonn, 2020-09-19:
I might be very blind, but this is the only thing I can find in the components list. https://gyazo.com/68ab40958666fc51024135cad76f2c57

olikraus, 2020-09-19:
Every example includes only a small subset of the available constructors.

https://github.com/olikraus/u8g2/blob/5e8960d931d41a2d3c38752f44d56839039a0402/sys/arduino/u8g2_page_buffer/HelloWorld/HelloWorld.ino#L74-L82

The full list of constructor is here:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

TheRealArtonn, 2020-09-19:
Success! or atleast some kind of success. There is text now, but no animations. https://gyazo.com/85fd1fe00e60e89a34a551f55c01bb1d

TheRealArtonn, 2020-09-19:
Extra note: using "U8G2_SH1107_64X128_F_SW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);" instead of "U8G2_SH1107_64X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);" seems to show more on the screen.

olikraus, 2020-09-19:
Software emulated is much slower, but any other pin could be used

TheRealArtonn, 2020-09-20:
Well I got the monitor working perfectly for me now, So I am happy! Thanks for the help!

closed 2020-09-21 by olikraus