olikraus/u8g2 #1288: UpdateDisplayArea character shift

periperiperiperi, 2020-09-19:
Hi, im new to this so apologies in advance. I seem to be having a similar issue to another user (#1277 ) where, sendbuffer and update display work perfectly, however when i use updatedisplayarea it appears as garbage. after investigating, it is actually shifting parts of the character along in proportion with the specificed tile width and height. If i specify the max height and only half the width, then only every second line of the word prints.  I am using atmega328P and st7920 128x64 blue/white lcd. this means that the max tile size is 16x8. examples shown below.
max height, arbitrary width gives garbage
[image]
[image]

max height, exactly half width gives every 2nd line (kind of).
[image]
[image]

max height max width gives expected output
[image]
[image]
Thanks so much

olikraus, 2020-09-19:
The quick answer is: UpdateDisplayArea is not supported for the following display controllers: SH1122, LD7032, ST7920, ST7986, LC7981, T6963, SED1330, RA8835, MAX7219, LS0xx
But it should mention this in the description.

periperiperiperi, 2020-09-19:
ah okay sorry about that, thanks anyway!
a quick workaround i managed to come up with was to update a slice of the screen. that is, update the full width and only a portion of the height. this allowed much higher refresh speeds

olikraus, 2020-09-19:
I have updated the menual.

https://github.com/olikraus/u8g2/wiki/u8g2reference#updatedisplayarea

closed 2020-09-26 by periperiperiperi