olikraus/u8g2 #1291: custom resolution 2040x12

bance, 2020-09-24:
Hi there. I have a custom led matrix that displays a buffer of 2040 by 12 pixels. What would be the easiest approach to get u8g2 support on such a resolution? 

no need to interface with actual hardware here. I have successfully read and displayed the internal buffer of a ssd1306 object but it's limited to 128 pixels max in width.

any hint will be greatly appreciated!

olikraus, 2020-09-24:
> it's limited to 128 pixels max in width.

I can just create a copy of the ssd1306 object and extend the size accordingly.
u8g2 16 bit mode would be obviously required (the #define needs to be set, see FAQ).

bance, 2020-09-24:
thanks for the quick reply! that would be greatly helpful if you wouldn't mind creating it, or let me know where to change. 

again, thanks for this great project.

olikraus, 2020-09-25:
The constructor is called "U8G2_SSD1306_2040X16_1_4W_SW_SPI". Other constructors for HW SPI also exist, including the 2 and F constructor.

Beta release for this is 2.28.7

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

bance, 2020-09-27:
Thanks Oli! will get a thorough test and get back to you.

bance, 2020-09-27:
Hi Oli, it appears that the content in frame buffer is wrapped at 256bytes. if, for example, I wrote 170 characters using drawUTF8 with a 12px font, only the last ~20 characters showing up at the beginning of the frame buffer overlapped with some random pixels that appeared from earlier characters.

bance, 2020-09-27:
Found the problem. it now works flawlessly after enabling U8G2_16BIT in u8g2.h.

Many thanks, Oli! now I could enjoy the many features provided by u8g2 with my existing and future customised LED matrix.

btw, guess it would be extremely useful for people making customised led matrix if there is a special constructor that allows user-defined resolution. in such a constructor there would be no need to support actual hardware - full frame buffer is all that matters.

All the best!

olikraus, 2020-09-27:
U8g2 avoids memory allocation. As a consequence the dimensions must be available at compile time. There can not be a constructor for dynamic dimensions.

bance, 2020-09-27:
Right. now I understand much better how to create a constructor with specific resolution by reading your code.

thanks for your generous help. I believe other makers will benefit from this as well.

closed 2020-09-27 by bance