olikraus/u8g2 #1292: st9720 to atmega328P

fedcard87, 2020-09-25:
Im trying to configure using your u8g2 setup for the st9720, but we are having an issue with noise being present on the LCD screen. it prints the word but only on a little bit of the top of the screen, and the rest is like noise. When doing the hardware setup, what exactly should the defined ports/pins from this be connected to for the SPI communication.
#define DISPLAY_CLK_DIR DDRB
#define DISPLAY_CLK_PORT PORTB
#define DISPLAY_CLK_PIN 5

#define DISPLAY_DATA_DIR DDRB
#define DISPLAY_DATA_PORT PORTB
#define DISPLAY_DATA_PIN 3

#define DISPLAY_CS_DIR DDRB
#define DISPLAY_CS_PORT PORTB
#define DISPLAY_CS_PIN 2

//#define DISPLAY_DC_DIR DDRB
//#define DISPLAY_DC_PORT PORTB
//#define DISPLAY_DC_PIN 1

#define DISPLAY_RESET_DIR DDRB
#define DISPLAY_RESET_PORT PORTB
#define DISPLAY_RESET_PIN 0

fedcard87, 2020-09-25:
we currently have configured the RST  to the reset_dir, cs to the cs_dir, RW to the data_dir and the enable pin to the CLK_dir on the LCD, but are unsure where the DC_dir should go to?

olikraus, 2020-09-25:
> Im trying to configure using your u8g2 setup for the st9720, but we are having an issue with noise being present on the LCD screen. 

Probably you use page mode without the required picture loop.
For full buffer mode, you need to use the "f" constructor.

closed 2022-03-04 by olikraus