olikraus/u8g2 #1294: Support for NHD-2.23-12832UCW3 SSD 1305 ?

nordseele, 2020-09-27:
Hi, 
Thanks for making this library.
Would it be possible to add support for the [NHD-2.23-12832UCW3](https://www.mouser.fr/datasheet/2/291/NHD_2_23_12832UCW3-1222125.pdf) OLED display please ? It's a 128x32 SSD1305 display.

It's not part of the list of supported devices already but I'm testing the wiring with the constructor of another device U8G2_SSD1305_128X32_ADAFRUIT_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); and the "page buffer, graphics test" example provided. (on a Teensy 3.6)

The bottom of some pages is cropped and the "transparent bitmap overlay" is a bit flickering (maybe that's normal, I don't know) but beside that it seems to work. I'd be happy to provide more details or to do some tests if necessary. 

Or should I use one of the existing constructor and try to adjust the offset or something ? Please let me know.

Thanks.

[image]

olikraus, 2020-10-03:
Can you run the page mode "FlipMode" example? If the frame at the display borders stays constant and never changes, then everything is fine.

nordseele, 2020-10-03:
Yes, sure.
The frame remains constant indeed (photo A). Some "ghost" pixels briefly appear/flash on each rotation (photo B).
I'm not sure wether it's the normal behavior since I'm not familiar with the library yet but I will do more tests now.
Thanks.

##### Photo A
[image]

##### Photo B
[image]

nordseele, 2020-10-03:
Just did a few more tests, trying various constructors (Full screen, Page buffer and HW and SW), nothing super intensive yet but I think everything's fine with this display. That's good news :)

Thank you very much!

olikraus, 2020-10-03:
Not sure from where those pixel came from, but I also assume, that everything looks good. Thanks for all the pictures.

closed 2021-05-09 by olikraus