olikraus/u8g2 #1295: can constructor name be a variable?

Kmanggaard, 2020-09-29:
My terminology is probably way off, but I'm trying to do something like this:

U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI disp0(U8G2_R0, /* cs=*/ 2, /* dc=*/ 10, /* reset=*/ 9);
myVar = disp0;
myVar.sendBuffer();

Hope that makes sense.
If it's possible, what kind of variable should I use?

olikraus, 2020-10-01:
It would be a waste of memory, but should work.
I acctually woruld prefer pointers instead.
``` C++
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI *myVar
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI disp0(U8G2_R0, /* cs=/ 2, / dc=/ 10, / reset=*/ 9);
myVar = &disp0;
myVar->sendBuffer();
```
But all in all it is more a C++ question...

Kmanggaard, 2020-10-04:
Memory I have plenty of, so I'll give it a go. Thanks for an incredible
library and great support.

olikraus, 2020-10-04:
> Memory I have plenty of, so I'll give it a go. 

Pointer are also faster. ;)

closed 2021-05-09 by olikraus