olikraus/u8g2 #1296: How to convert unicode to .bdf?

mianqi2016, 2020-09-30:
I have read this issue:
@olikraus / How to customize my own set of Chinese characters to show in LCD? #510
and I have converted these words "前方地滑，小心脚下！" into unicode, which were "U+524D U+65B9 U+5730 U+6ED1 U+FF0C U+5C0F U+5FC3 U+811A U+4E0B U+200B U+FF01 U+200B"

my question is how to convert it into .bdf file?

mianqi2016, 2020-09-30:
I found some clues here:
"Here is the file for the gb2312 font: https://github.com/olikraus/u8g2/blob/master/tools/font/build/gb2312.map
You can download gb2312.map and regenerate the font with -M gb2312.map
You could also remove glyphs which you do not need from gb2312.map and generate a smaller version of the font."

that means I could download the .map file, then delete the words I need not, just keep the very limited ones I want, then use the bdfconv.exe tool regenerate the file. But I couldn't open the links for "gb2312.map" and "bdfconv.exe", could anyone here send me the 2 files? my mail:mian.qi@outlook.com

olikraus, 2020-09-30:
Basically you can use any map file.
Another map file is here: https://github.com/olikraus/u8g2/blob/master/tools/font/build/chinese1.map

The converter itself is here: https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/bdfconv.exe

mianqi2016, 2020-09-30:
Can I make my own .map file? Since I couldn't open the 2 links.

https://postimg.cc/GBNdXWPJ

mianqi2016, 2020-09-30:
> The syntax of the map file is the same as the "-m" command:

What does that mean? Does it mean I could generate a map file by a txt file via bdfconv.exe with "-m" arguments?

mianqi2016, 2020-09-30:
While, I still need a .bdf file? It this true? If it is, how could I make my own .bdf file for just several tens Chinese words?

And, what's the difference between .bdf file and .map file?

Should I open them in other software, or I could read them in .txt model?

olikraus, 2020-09-30:
> What does that mean? Does it mean I could generate a map file by a txt file via bdfconv.exe with "-m" arguments?
Yes, the map file is a simple text file.
And no, -m will not generate anything, it will just expect the same syntax like what you need to use in the map file.

> While, I still need a .bdf file? It this true?

bdf can be exported from fontforge tool. You can also use a tool called fony to export to bdf. And finally there is also a ttf to bdf converter called otf2bdf.

> Should I open them in other software, or I could read them in .txt model?

map and bdf are both pure text files. You can use an editor with both.

> And, what's the difference between .bdf file and .map file?

A bdf file contains the bitmap image of each glyph (e.g. kanji). The map file or the -m command selects a specific subset of those glyphs. A bdf file with all far east glyphs will be very huge and usually you need only a small number of them. This is why we have the map file. The map file selects those glyphs which you really need in your project. You can keep the bdf file untouched or take it from some eexisting sources. 

For example you can get bdf files (and map files) from here: https://github.com/larryli/u8g2_wqy

mianqi2016, 2020-10-03:
I asked the author of Fony, it was told, the tool can't convert a .txt file into a .bdf file, so if I want my own map file, as to say, a subset of a bdf file in .txt format(since I would use your tool bdfconv .exe convert it into code in C program), how to do it?

mianqi2016, 2020-10-03:
WQY gave a solution here: https://github.com/larryli/u8g2_wqy/wiki/CustomFont

echo '32-128,' > myfont.map
cat str.txt | iconv -f utf-8 -t c99 | sed 's/\\u\([0-9a-f]\{4\}\)/\$\1,\n/g' | sort | uniq | sed '/^$/d' | tr '/a-f/' '/A-F/' >> myfont.map

but "cat" couldn't run on windows 10.

[image]

mianqi2016, 2020-10-03:
Maybe I need to try it on Ubuntu, will bdfconv run on Ubuntu?

olikraus, 2020-10-03:
I do everything on linux only. You just need to download/clone the u8g2 repo and run the make command in the bdfconv directory. this will build the ubuntu executable. Maybe you need to install gcc before this.

vincent113, 2020-11-10:
build your own font .c file and use it.
need .bdf font file, unicode .map file and bdfconv.exe.

i had try build this on windows platform with vscode work fine.

good luck.

[image]

mianqi2016, 2020-12-22:
I downloaded the latest u8g2-master.zip this morning, and tested it in Arduino IDE, it's OK.

Schön, vielen vielen Dank an Sie, Olikraus.

closed 2020-12-22 by mianqi2016