olikraus/u8g2 #1297: Proteus simulation issues

valfa14, 2020-09-30:
https://github.com/olikraus/u8glib/issues/513
I'm facing also same issues with u8g2 lib :/

olikraus, 2020-10-01:
I think I can not help here. I do not know anything about "proteus simulator". 
U8g2 works in Arduino IDE as expected. If it doesn't work in "proteus simulator" then I would suggest to contact "proteus simulator" people.

SWIPHER, 2021-07-29:
Hi,I had the same problem with you but I solved just now.I found it only works with SPI mode in Proteus, the code is like this
`U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0,24,25,26,23,27);`
It works on the Arduino mega2560 moudle in proteus. And the Circuit diagram is like this:
[image]

closed 2022-03-04 by olikraus

223222981, 2022-05-14:
> 
[image]

Thank you I succeeded too