olikraus/u8g2 #1298: St7571 Display

s54b32julian, 2020-10-02:
Hello Oli,

I don’t see 128x128 ST7571 in constructor. Is it available via other constructor or not supported at the moment?

Best regards,

olikraus, 2020-10-03:
If it is not in this list: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
then it is not supported. :-(

s54b32julian, 2020-10-03:
Would it be possible for you to add this with this datasheet https://www.crystalfontz.com/controllers/Sitronix/ST7571/305/ or if I send you one display? There are sadly no other displays in 128x128pixel with ~45x45mm display area.

s54b32julian, 2020-10-14:
No chance? The display is not that expensive, but of course would come free to you ;) maybe it‘s not much work, as they are very similar to UC1611. And maybe to the others from ST-Drivers.

olikraus, 2020-10-18:
The problem is mainly my available time on working on this. Unfortunately, the ST7571 differs a lot from UC1611. 
It would be easier to have a specific init sequence for your display. Is there any datasheet for your display (you have provided the datasheet of the controller)?

olikraus, 2020-10-18:
I have ordered this display: https://aliexpress.com/item/32608068279.html
Will arrive in 4 weeks.

s54b32julian, 2020-10-21:
Oli you are the real MVP. Is there a way to spend you a coffee or beer? Some people here on Github have a paypal-link for that, you too?

olikraus, 2020-10-26:
No donation required, if any, I would take the beer :-D 
Let's wait until I receive this display. My e-mail (see any source code) is connected to paypal.

olikraus, 2020-12-27:
Added support for this display.

[image]

closed 2020-12-27 by olikraus