olikraus/u8g2 #1299: Incorrect graphics on SSD1322 based ER-OLEDM032 display

jrs-tech, 2020-10-02:
I have some problems using this library on SSD1322 based 256x64 display.
My display 'half' works - literally :D
Half of the display displays graphics, while other half is blank most of the time - i get a few pixels on the blank side on some u8glib demo pages, it looks like parts of geometrical primitives get displayed in the wrong place. 
Initially i though it is a HW problem, but display works fine when i use the library and demo code supplied by the seller, which can be found here:
https://www.buydisplay.com/white-3-2-inch-arduino-raspberry-pi-oled-display-module-256x64-spi

I use 'U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI' constructor, it is on mega2560

Any suggestions welcome!

olikraus, 2020-10-03:
Locate file u8x8_d_ssd1322 on your hard disc. Try to locate the following line:

https://github.com/olikraus/u8g2/blob/cd1027c84e3f683b9c8bf507cb864e5dd2702c98/csrc/u8x8_d_ssd1322.c#L278

Change the second argument to something else than 0x00. Will this change the output?

jrs-tech, 2020-10-03:
I have missed the '// Enable U8G2_16BIT in u8g2.h' 
All works fine after enabling it

closed 2020-10-03 by jrs-tech

olikraus, 2020-10-04:
oh, great then...