olikraus/u8g2 #1301: Attempting to Create my own Minimalistic U8x8 font but it Won't Display Properly

itsbrockwillie, 2020-10-04:
Hello!
I've been using your amazing library in my recent project but I've hit a bump in the road, the fonts don't fit on the small amount of flash the Arduino Pro Mini's I have contain. I'm trying to make a 4 character long u8x8 font, I've cheated a little by using letters A, B, C and D as N, M, V and 1. I converted the custom .bdf from Fony to the u8x8 .c file type with the command "bdfconv.exe -v -f 2 -m "65-68" NMV1.bdf -o NMV1.c -n NMV1" (file is attached .txt file copy and pasted)

[NMV1.txt](https://github.com/olikraus/u8g2/files/5324560/NMV1.txt)

I added this file to "Arduino\libraries\U8g2\src\clib\u8x8_fonts.c" and added "extern const uint8_t NMV1[] U8X8_FONT_SECTION("NMV1");" to "Arduino\libraries\U8g2\src\clib\u8x8.h"

I added "u8x8.setFont(NMV1);" and "u8x8.drawString(0,1,"ABCD");" to my code only for my OLED to display this: (sorry it's dark)

[image]

I suspect it has something to do with size ratios as the characters displayed are 6x8 pixels big and I can see the start of the V in the third letter. The actual font is supposed to be 48 pixels tall so this ratio displayed is far from what I need. Any and all help is appetited.
Thank you!

olikraus, 2020-10-05:
Can you also share the bdf file?

itsbrockwillie, 2020-10-05:
Sure thing, it won't let me just paste the file so here's the google drive link:
https://drive.google.com/file/d/1aq5B1VjDNmFU8GuZx2Bpiv9IivAUJTAd/view?usp=sharing

olikraus, 2020-10-18:
Ah i see, the glyphs are 16x16. You need to use the options "-th 2 -tv 2"

itsbrockwillie, 2020-10-20:
This is on the right path as it did show more of the glyphs but they are still too small. I'm pretty sure they are actually 48x48. Do you know what settings I would use in that case instead? Thanks!

olikraus, 2020-10-26:
In this case, try "-th 6 -tv 6".

itsbrockwillie, 2020-10-29:
Perfect, this helped. It did originally cause the characters to have huge spaces between them but after adjusting "-th 6 -tv 6" to "-th 3 -th 6" it all worked out. In case anyone has this same problem and they're reading this, just note "-th" controls the width while "-tv" controls the height. Thank you so much for the help!

closed 2020-10-29 by itsbrockwillie

AnHardt, 2020-10-30:
-t**v** -> **v**ertical.
-t**h** -> **h**orizontal.