olikraus/u8g2 #1307: Trying to port the library to STM32, no go

spi5498, 2020-10-09:
So after days of puzzling bits of information together, i came up with this:
[https://pastebin.com/L75pGi92](url)

I have checked that i have "traffic" on my scope. Its only two channels, so i have checked it a bit rough. Ie CS low when data is transferred and so. However nothing comes up on the LCD. The wiring is checked 10 times, atleast.

Does anyone see anything obvious? Otherwise i need to dig out my logic analyzer, however it does not have SPI decoding.

spi5498, 2020-10-10:
Did notice that i had "disable" in the MSG_BYTE_INIT, changed that to enable instead ofcourse. Same problem though..

spi5498, 2020-10-10:
So, after more or less two days wasted, still cant get any life out of this with the STM. I have no idea why it is so hard.
Attaching a snippet from the scope. this is from the init routine.
Yellow: RST
Red: CLK
Blue: CS
Green: Data

[image]

spi5498, 2020-10-10:
I did try out the SPI decoding on my PicoScope, and with just the code for Init and powersave i get this over the SPI:

0 02 32 50 C1 6C 00 87 00

Does that make any sense?

spi5498, 2020-10-10:
Nope, data is totally wrong.

spi5498, 2020-10-10:
I see that there is 30 commands sent in the init sequence and thats corresponds to whats in the sourcefile for my controller. 
Atleast thats something.
However, it is completely the wrong data.

closed 2020-10-11 by spi5498

olikraus, 2020-10-11:
According to your code https://pastebin.com/L75pGi92 , i see the following issues:
1) the gpio&delay procedure should implement all the delay messages.
2) I wonder that there is no RS/A0/DC handling

spi5498, 2020-10-11:
Thanks for your response, as i said, this code that is posted here will not be used anymore. I will do a complete makeover.
the reset procedure in the code for my display controller is commented out. Is that not used?

olikraus, 2020-10-12:
> the reset procedure in the code for my display controller is commented out. Is that not used?

Do you have link to the code?

olikraus, 2020-10-12:
oh, same question in new post. I will answer there.