olikraus/u8g2 #1308: STM32 and ST7586

spi5498, 2020-10-11:
I have a couple of quick questions to be able to port this to a STM32 controller

Firstly, my lcd is the st7586s_erc240160 and from what i can see, the only call to RESET is in the init, and that one is commented out.

By this, can i safely disregard in writing any code for the reset? And just leave that at level 1?

Also, regarding some timing info. The STM32 seems a bit problematic when doing delays that is below 1ms.

Its ofcourse possible, but for now, would you say the wait for CS would be ok with 1ms? I see that there is a couple of ns references in the display info section of your code.

I just wanted to wait with those routine and keep it as simple as possible until i get some output at all from the LCD.

Thank you

Link to datasheet: [https://www.buydisplay.com/download/ic/ST7586.pdf](url)

olikraus, 2020-10-12:
> Firstly, my lcd is the st7586s_erc240160 and from what i can see, the only call to RESET is in the init, and that one is commented out.

You mean this line?
https://github.com/olikraus/u8g2/blob/cd1027c84e3f683b9c8bf507cb864e5dd2702c98/csrc/u8x8_d_st7586s_erc240160.c#L83

I think this can be ignored. In fact, if I remember correctly, the code was contributed. I do not know whether this device works as expected.

> By this, can i safely disregard in writing any code for the reset? And just leave that at level 1?

Reset code will be handled and is common for all displays, it is handled here:
https://github.com/olikraus/u8g2/blob/cd1027c84e3f683b9c8bf507cb864e5dd2702c98/csrc/u8x8_display.c#L64-L77

> Its ofcourse possible, but for now, would you say the wait for CS would be ok with 1ms? I see that there is a couple of ns references in the display info section of your code.

Yes, it is always ok to use larger delays.

spi5498, 2020-10-12:
Thanks!

I have everything up and running now. So i will just double check all the timings with the scope, and then comment the code as much as possible. Maybe we can post it here, for future reference?

closed 2022-03-04 by olikraus