olikraus/u8g2 #1309: SSD1322 256x64 only using 240x64

nbelanger99, 2020-10-14:
Hi, I have a Newhaven 256x64 SSD1322 display. When using the U8G2_SSD1322_NHD_256X64_1_4W_HW_SPI constructor, the test code works fine, and I can draw things to the screen and all, but the screen size is incorrect; it's only using 240x64 of the pixels (there's a block on the right that's unused). Any idea of how I can fix this?

v1993, 2020-10-14:
You might need 16-bit mode. See [FAQ](https://github.com/olikraus/u8g2/blob/master/doc/faq.txt) on how to.

nbelanger99, 2020-10-15:
> You might need 16-bit mode. See [FAQ](https://github.com/olikraus/u8g2/blob/master/doc/faq.txt) on how to.

This fixed the issue, thanks! That could be a little more clear in the docs.

olikraus, 2020-10-18:
It is also there in the manual: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#16-bit-mode
I also tried to mention this in the Arduino examples: 
https://github.com/olikraus/u8g2/blob/cd1027c84e3f683b9c8bf507cb864e5dd2702c98/tools/inoupdate/page_buffer.ino#L32

closed 2020-10-18 by olikraus