olikraus/u8g2 #1311: u8x8 library

Pacimani, 2020-10-15:
Hi Oli!
I am trying to find the set-up for my LCD "ST7920 128X64" but I can't find it under u8x8 library documentation. Is there any suggestion for the set-up of this type of LCD?

Thanks

olikraus, 2020-10-15:
Yes, it's not there. Due to the internal memory architecture of the st7920, u8x8 will not work.

Pacimani, 2020-10-16:
Thanks for your prompt response. I am using the hardware SPI as advised, and for some reason that we are yet to find, it's slowing down the clock speed of other processes that we are running on the same device. I have been looking into a lightweight c library that supports the changing the orientation of the LCD screen but I couldn't find any.

olikraus, 2020-10-26:
With u8g2 the orientation can be changed via constructor argument. Lightweight: There are some #defines in u8g2.h to reduce code size. Also I suggest to look into FAQ: https://github.com/olikraus/u8g2/blob/cd1027c84e3f683b9c8bf507cb864e5dd2702c98/doc/faq.txt#L171

closed 2022-03-04 by olikraus