olikraus/u8g2 #1312: esp32 with 19264 ks0108 lcd miss the pixels position with every redraw

GoDotMad, 2020-10-16:
this is the hello world example 

I tried it in page buffer (1 & 2 page) and in full buffer 
as well as in u8x8 mode
it seems that when a redraw of the buffer occurs some pixels shift or something

GoDotMad, 2020-10-16:
[image]

GoDotMad, 2020-10-16:
[Uploading VID-20201016-WA0001.mp4…]()

GoDotMad, 2020-10-16:
[image]

GoDotMad, 2020-10-16:
https://gifyu.com/image/8w6i

GoDotMad, 2020-10-16:
forgot to mention that it works without any problems using arduino uno 

so is it possible that it is caused by the difference in processor architecture 

also I'm using platformio on vscode not arduino ide

GoDotMad, 2020-10-18:
terns out that my lcd needs a little bit more time after the enable signal 

I tried to adjust the times in the screen info struct
but it was already near maximum with 250 ns
so I adjusted the u8x8_byte_ks0108 function by coping the call to the delay function once more

I think that the reason it worked on the uno is because it is slower than the esp32 

and by the way 
this is a super cool library 
every thing is organized and modular 
and all you need to modify it is some time and understanding of the lcd controller

so thanks for this great library

closed 2020-10-18 by GoDotMad

olikraus, 2020-10-18:
Excause my delay in looking into issues. Just read your posts.
I am happy you solved this issue. Good work.

AhmadSarifudin, 2021-11-01:
@GoDotMad Hi 

> terns out that my lcd needs a little bit more time after the enable signal
>
> I tried to adjust the times in the screen info struct but it was already near maximum with 250 ns so I adjusted the [...]

Hi @GoDotMad  , I have same problem , Can you explain  where part do you change about times in the screen info , in the library? , Thanks

GoDotMad, 2021-11-01:
Hi @AhmadSarifudin ,

are you usind a screen with the same controller as the one I mentioned?

flewis88, 2025-01-29:
@GoDotMad I'd be interested to know your workaround too. I'm getting broken graphics

GoDotMad, 2025-03-21:
Hi there, 
what lcd and what controller are you using? 

I hope I'm not too late to reply.

flewis88, 2025-03-31:
ESP32 is generic, similar to this: https://joy-it.net/en/products/SBC-NodeMCU-ESP32
And the KS0108 is model ABG128064A50-DIW-R

Appreciate you replying!