olikraus/u8g2 #1314: KS0713 based display

Fabrice82, 2020-10-16:
Hi !
I'm using a 64x128 KS0713 based display. I've read the 2 previous issue opened about this display, but seems to not be resolved.
I've used this display with the old U8Glib, with this constructor :
 U8GLIB_LM6059_2X     display (LCD_CS, LCD_DC, LCD_RESET);
It works almost fine : display is just mirrored - no matter for graphic use.

Now I want to display text and U8Glib does not have mirror option.

I haven't found any constructor in the U8G2LIB able to work with this display (same setup as for U8Glib, which is working almost fine !) -> not an HW issue
There is a loss of compatibility, mainly with this constructor I guess:
U8G2_ST7565_LM6059_2_4W_HW_SPI display(U8G2_R0, LCD_CS, LCD_DC, LCD_RESET); 

Any idea or turnaround ?

Thanks
Fabrice

olikraus, 2020-10-18:
I am confused. It should actually work. Did you try the ContrastTest example?
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/ContrastTest/ContrastTest.ino

Will you see something? Maybe the contrast differs between u8g and u8g2.

Fabrice82, 2020-10-18:
Test done -> no success ...
I've already think about that possible issue, and tested with contrast = 0x10 (works with U8Glib), then 0x127 on U8G2lib, without any difference.

I'm using a "3V 12MHz trinket pro" board. A test LED is present and connected to the SPI-CLK, so I see that something is alive on the SPI bus ! As I say, same HW setup between U8G and U8G2.

Also tried the SW_SPI constructor. Still nothing on the screen.

olikraus, 2020-10-19:
Strange. So u8glib works with the same setup?
Maybe still the u8g2 constructor args are wrong?

Fabrice82, 2020-10-20:
Same setup at all and the code include the two libraries with a '#define' to select it:

`#define use_U8G
#ifdef use_U8G
   #include <U8glib.h>
#else
   #include <U8g2lib.h>
#endif

#ifdef use_U8G
   //U8GLIB_LM6059_2X display (LCD_CS, LCD_DC, LCD_RESET);
   //U8GLIB_LM6059_2X display (LCD_CS, LCD_DC);
#else
   //U8G2_ST7565_LM6059_1_4W_HW_SPI display(U8G2_MIRROR, LCD_CS, LCD_DC, LCD_RESET);
   U8G2_ST7565_LM6059_1_4W_HW_SPI display(U8G2_MIRROR, LCD_CS, LCD_DC);
#endif
`
The loop just screen up a "hello world!" ! I cannot do more simple .. Run mirrored with U8G / don't run with U8G2. 
I've tested all constructor one by one, removed the RESET arg, swapped DC and CS ...
Nothing seems to have any effect.

Fabrice82, 2020-10-21:
well,
I've completely review the init sequence:
static const uint8_t u8x8_d_st7565_lm6059_init_seq[] = {
  U8X8_START_TRANSFER(),             /* enable chip, delay is part of the transfer start */
  U8X8_C(0x0ae),		                /* display off */
  U8X8_C(0x0e2),            			 /* soft reset */
  U8X8_C(0x0a3),		                /* LCD bias 1/9 */
  U8X8_C(0x0a0),		                /* ADC set to reverse */
  U8X8_C(0x0c0),		                /* common output mode */
  U8X8_C(0x02f),		                /* all power  control circuits on (regulator, booster and follower) */
  U8X8_C(0x026),		                /* set V0 voltage resistor ratio to max  */
  U8X8_C(0x0a6),		                /* display normal, bit val 0: LCD pixel off. */
  U8X8_CA(0x081, 0x010),		       /* set contrast, contrast value, EA default: 0x016 */
  U8X8_C(0x0af),		                /* display on */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};
Now it run right !
Mirror issue corrected also !
Hope it will help other people with this display KS0713 based chip...

olikraus, 2020-10-26:
I have created a new constructor U8G2_ST7565_KS0713_xxx with your updated sequence.

I have also created a new beta release 2.28.8

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2020-10-26:
Maybe you can give some feedback on the new constructor whether this works or not.

closed 2020-10-26 by olikraus