olikraus/u8g2 #1315: ST75256 100x160 LCD display

BenW5122, 2020-10-20:
Hi, 

I am trying to run the example codes you provided, but im using a 100x160 LCD display. 
May i know what changes do i need to make to make my display work ?

olikraus, 2020-10-26:
The ST75256 is already supported, but not your specific dimensions. The first step is to identify an existing constructor, which works correctly. Use the flip mode example and test the some of them: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st75256-jlx256128

In probably all cases you will either see no pixel or all pixel enabled. This is due to the very small contrast window of the ST75256 display, so you need to adjust the contrast first (setContrast command).
Maybe start with "U8G2_ST75256_JLX172104_1_4W_SW_SPI, figure out the contrast value (maybe with help of contrast example programm) and check whether the text is displayed correctly.
If the text is not displayed correctly try a different constructor.

You could also post pictures here.

olikraus, 2022-03-04:
no further response, closing...

closed 2022-03-04 by olikraus