olikraus/u8g2 #1317: Hardware SPI AVR

fedcard87, 2020-10-24:
Im trying to implement hardware spi for my lcd being a st7920, using an atmega328P chip. Is there an example code for hw spi that uses avr and not arduino, thanks.

olikraus, 2020-10-26:
Did you check the porting guide?
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

happymacer, 2021-09-12:
> Im trying to implement hardware spi for my lcd being a st7920, using an atmega328P chip. Is there an example code for hw spi that uses avr and not arduino, thanks.

Ive done some work on that but using i2c.  In Peter Fleury's i2cmaster library (included with u8g2) he refers to "AVR_USE_HW_SPI" suggesting this library could be used just like the i2c library (see the u8x8_avr.c file).  You can try it, and also have a look at my video to set up u8g2 on AS7 on a barebones Atmega328p with i2c - https://youtu.be/TO7-yxT6Gvw if you get stuck.

closed 2021-11-19 by olikraus