olikraus/u8g2 #1318: SH1122 F 4W HW SPI only shows half of the images

Jack-XHP, 2020-10-24:
I used to run a 256*64 SH1122 in i2c and it works perfectly, just a bit slow. So I decide to run it in SPI, however, after I connect SCK to pin 52, data to 51, cs to 10, dc to 9 and res to 8 on my mega 2560. The display only shows half of the characters.
[image]

```
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_SH1122_256X64_F_4W_HW_SPI u8g4(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
void disStr256(char* text){
  u8g4.clearBuffer(); 
  u8g4.setCursor(0, 64);
  u8g4.print(text);
  u8g4.sendBuffer();
}
void setup() {
  // put your setup code here, to run once:
  Wire.begin();
  u8g4.begin();
  u8g4.setFont( u8g2_font_logisoso32_tf);
  disStr256("test");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

v1993, 2020-10-24:
See FAQ on how to enable 16-bit mode.

olikraus, 2020-10-26:
Will this solve your problem?

theeprawn, 2020-10-26:
For SPI, I use would also replace Wire.begin(); with SPI.begin();
I'm not geared up to test with Atmel to test atm, but worked with u8g2 with I2C and SPI 3W (saves a I/O pin) on SH1122(256x64) and yes I found I2C to slow with 4bits per pixel for SH1122, max I could reach was 4fps, with SPI I get over 35fps.

closed 2022-03-04 by olikraus