olikraus/u8g2 #1319: psoc4 hardware SPI 1306screen, clock issue

BLLTeam, 2020-10-25:
I try to port the great u8g2 librairy in a Psoc projet with SPI.

I use SPI 2.50 component.

For now, it's not ok. I compare the same screen but with arduino hardware SPI, I can see that some SPI clock are missing.

Is there someting wrong into my two specific callback function ?

#include <project.h>
#include <stdlib.h>
#include <stdio.h>
#include <assert.h>

#include "u8g2.h"

/* 
 * A PSoC SPI HW Hardware Interface for U8X8
*/

    uint8_t psoc4_BLL_gpio_and_delay( u8x8_t *u8x8, uint8_t msg,  uint8_t arg_int,void *arg_ptr)
    {
        (void)u8x8;
        (void)arg_ptr;
        switch (msg)
        {
          case U8X8_MSG_GPIO_AND_DELAY_INIT:
            break;
          case U8X8_MSG_DELAY_MILLI:
            CyDelay(arg_int);
            break;
          case U8X8_MSG_DELAY_10MICRO:
            CyDelayUs(10);
            break;
          case U8X8_MSG_DELAY_100NANO:
            CyDelayCycles(5);
            break;
            
        }
        
     return 1;   
    }
    
    
    uint8_t u8x8_byte_4wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int,
    void *arg_ptr)
    {
        switch (msg)
        {
            case U8X8_MSG_BYTE_SEND:
                //SPIM_WriteTxData(arg_int);
                SPIM_PutArray(arg_ptr, arg_int);
                break;
            case U8X8_MSG_BYTE_INIT:
            
                break;
            case U8X8_MSG_BYTE_SET_DC:
                // switch from cmd (arg_val = 0) to data mode (arg_val = 1) or vice versa
                // make sure SPI FIFO is empty before changing A0. Otherwise, the previously applied A0
                 // (valid for all queued SPI transfers so far) will change before the corresponding SPI data is clocked out
                
                while((SPIM_ReadTxStatus() & SPIM_STS_SPI_IDLE) == 0)
                {
                  // wait for SPI to finish
                }
                
                A0_Write(arg_int);
                
                //u8x8_gpio_SetDC(u8x8, arg_int);
                break;
            case U8X8_MSG_BYTE_START_TRANSFER:
                
                break;
            case U8X8_MSG_BYTE_END_TRANSFER:
                break;
            default:
            return 0;
        }
        return 1;
    }

int main()
{
  
  // enable global interrupts
  CyGlobalIntEnable; 
    
    SPIM_Start();
    
    
   
  // initialize display stuff...

    
    u8g2_t u8g2;
    
   
    
    u8g2_Setup_ssd1306_128x64_noname_1(&u8g2, U8G2_R0, u8x8_byte_4wire_hw_spi, psoc4_BLL_gpio_and_delay);

    
    u8g2_InitDisplay(&u8g2);// transfer init sequence to the display, display is in sleep mode after this,
   

    
    u8g2_SetPowerSave(&u8g2, 0);// turn on display

    
    u8g2_SetFont(&u8g2, u8g2_font_ncenB08_tr);
        u8g2_SetFontRefHeightExtendedText(&u8g2);
        u8g2_SetFontPosTop(&u8g2);

    
   
  for(;;)
  { 
   	u8g2_FirstPage(&u8g2);
    
  	do
  	{
  		
        u8g2_DrawStr(&u8g2, 0, 0, "Hello!\0");
        
  	} while( u8g2_NextPage(&u8g2) );
    
    
  

    CyDelay(1000);
    
  
    
  }
  
}

/* [] END OF FILE */

olikraus, 2020-10-26:
Hmm... you code does not change the CS line.

BLLTeam, 2020-11-06:
The CS line is change by the SPI hardware( when FIFO is empty)

olikraus, 2021-05-09:
not sure how I can help here :-/

closed 2022-03-04 by olikraus