olikraus/u8g2 #1323: ESP32 with HW i2c crashes during interrupt usage

andreash-esp, 2020-10-27:
Hi,
I hope this is the right place to ask.

My code crashes on my ESP32 (Heltec Wifi Kit 32) with integrated i2c OLED display when using HW i2c and counting a rotary encoder through interrupts. Using SW i2c works fine, but is obviously slower (170ms vs 35ms).
The crash occurs during a call to u8g2.sendBuffer();

I have tried various ideas to work around it with no success:
- disabling interrupts before calling sendBuffer()
- detaching the interrupt before calling sendBuffer()
- running the interrupt on Core 0

my stripped down code looks like this:
#include <U8g2lib.h>

//U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 15, /* data=*/ 4, /* reset=*/ 16);
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 16, /* clock=*/ 15, /* data=*/ 4);

long position = 0;
bool rotating;

void IRAM_ATTR isr2() {
  static int aState;
  static int aLastState = 2;
  
  if ( rotating ) delay (1);  // wait a little until the bouncing is done
   
  aState = digitalRead(outputA); // Reads the "current" state of the outputA
    if (aState != aLastState) {     
      if (digitalRead(outputB) != aState) { // counter-clockwise
         position--;
      } else {                              // clockwise
         position++;
      }
      rotating = false;
    }
    aLastState = aState; // Updates the previous state of the outputA with the current state
}

void setup() {
  // Rotary Encoder on pins 33 and 26
  pinMode(33,INPUT);
  pinMode(26,INPUT);
  attachInterrupt(33, isr2, CHANGE);

 // Init Display
  u8g2.begin();
  u8g2.setFont(u8g2_font_courB24_tf);
}

void loop() {
  rotating = true;

  u8g2.clearBuffer();
  u8g2.setCursor(10, 30);  u8g2.print(position);
  u8g2.sendBuffer();
}

This crashes the code immediately when the rotary is moved:
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Adding a "delay(100)" to the loop allows some input. It appears the code crashes when one or more interrupts arrive during sendBuffer().

Any idea where to look at?

Regards
Andreas

AnHardt, 2020-10-28:
A `delay()` in a ISR is not a good idea.

andreash-esp, 2020-10-28:
@AnHardt 
Thanks for your feedback. Both points are valid, and in fact my production code declares the position variable as volatile. I missed that when creating the test sketch. This shouldn't trigger the crash though.
As for the delay, I could not find a better way to debounce the mechanical rotary switch (KY-040).
I will try to remove it for testing to eliminate this possible issue.

olikraus, 2020-11-28:
Closing this issue, feel free to reopen for further questions.

closed 2020-11-28 by olikraus

andreash-esp, 2020-11-29:
Sorry for not coming back to this. I have since changed my Rotary Encoder to the ESP32Encoder library, which uses a pulse counter instead of individual interrupts for couting the ticks. This has fixed the issue.