olikraus/u8g2 #1324: Problem with drawStr function

EmriBG13, 2020-10-28:
Hi,
When i'm using the drawStr function, for some reason, I cant write below the y line in ratio to the font size.
for example:

void startup(){
  u8g2.clearBuffer();          // clear the internal memory
  u8g2.setFont(u8g2_font_6x10_tf);  // choose a suitable font
  u8g2.drawStr(30,7,"STARTING...");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  /*
   *  Put the SYSRESET* code here so the addresses will load in the background
   */

  delay(1500);
  u8g2.clearBuffer(); 
  u8g2.setFont(u8g2_font_6x10_tf);  // choose a suitable font
  u8g2.drawStr(25,7,"Select Mode:");  // write something to the internal memory
  u8g2.drawStr(0,21,"(A) GUI");  // write something to the internal memory
  u8g2.drawStr(0,28,"(B) Project Mode");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  return;
  
}
this is a function called to display a selection list (since using the buttons did not worked for me) but it doesn't display the text beyond the first line. 
Any help would be appreciated!

[image]

olikraus, 2020-10-28:
Did you use full buffer mode?

v1993, 2020-10-28:
(Please use ``` to format your code)

EmriBG13, 2020-10-28:
v1993 : OK I will use it next time.
olikraus : yes, I will change it to another modes and try again.

EmriBG13, 2020-10-28:
olikraus, I've updated the buffer to Full mode, for some reason I thought it was on F mode already, thanks!

closed 2020-10-28 by EmriBG13