olikraus/u8g2 #1326: ESP32 and ST7565 EA_DOGM128

ravimiranda, 2020-11-01:
Hi Oliver, 

Thank you for the excellent library and the notes. Apologies if this isn't the correct place to ask this query, but I'm running out of options at the moment.

I have two different DOG displays but I have been unable to make both of these work. Currently, I am trying the ST7565R EA_DOGL128-6 display with an ESP32-WROOM-32U Dev. board. I have tried various constructors under ST7565 EA_DOGM128, however, I am facing problems in getting this to work. 

I have wired the the display as in here: 
https://www.stoweb.net/?m_main=engineering&m_engineering=displaydogl&m_displaydogl=hardware

Can someone point me in the right direction if this has been done before, or, which pins in the ESP32-WROOM-32U Dev board would be the correct ones to use please?

If this isn't the correct place for the question, could I please be directed to the correct location.

Thank you for your time.

/R

ravimiranda, 2020-11-01:
Hi, 
An update, I managed to get the following construct working; U8G2_ST7565_EA_DOGM128_F_4W_SW_SPI u8g2
But, I can't seem to get the HW version of this working. Any ideas?

Thanks
/R

olikraus, 2020-11-28:
Usually HW pins are fixed and defined by your board. You need to use the correct pins as specified by your board and connect the SPI pins to those pins.
Maybe also U8g2 pin remapping works here. The HW SPI constructor accepts two more arguments to specify the target pins.

closed 2020-12-25 by olikraus