olikraus/u8g2 #1330: Conflict with other HW I2C slave

Max13, 2020-11-08:
Hi there,

I tried searching and reading many related questions and couldn't find an answer to my question. I'm on ESP32 (Lilygo T-Call with SIM800L modem), a 128x64 OLED I2C (`0x3C`) and an SHT35 I2C sensor (`0x44`).

I'm using `arduino-sht` library for the sensor and `u8g2` for the display. Both find their device on their own, separately. When I use them both together, the display works but not the SHT35, whether I put `sht.begin()` before or after `u8g2.begin()` (`U8G2_SH1106_128X64_NONAME_F_HW_I2C`)

Here is my working code with the sensor only output: https://gist.github.com/Max13/b1da7c4a8fb83fa1eb0e0b73b3ada85b/73956f4296990f2167793dfd328ad8919472e84e

```
> Serial ready
> Sensor: OK
> Probe: 21.35°C / 57.42%
// [...]
```

Here is when I add `u8g2` code: https://gist.github.com/Max13/b1da7c4a8fb83fa1eb0e0b73b3ada85b/bedd4ed32426958c7a6c8f0a7df3e666f39d1d2a

```
> Serial ready
> OLED ready
> Sensor: Error
> Error with probe
> Error with probe
```

Plus, the sensor lib "probes" the 2 common addresses and if I add a debug serial, I see that it tries both addresses and doesn't get successful reply so returns `false`.

Is there anything I can check with `u8g2` before switching to another temperature sensor lib maybe?

Thanks for your help

v1993, 2020-11-09:
Don't quote me on this, but I think it's arduino I2C interface limitation, generally not liking few devices on a single bus. Tweaking I2C speed may fix things (and slow down display), but really you should try native ESP-IDF instead of arduino wrapper. I've even ported u8g2 to it, but it is a bit out of date (C part will work with newer versions fine, C++ won't).

Max13, 2020-11-09:
Thanks for your reply @v1993, so in your opinion, I should either switch to ESP-IDF or use a software I2C for either modules?

olikraus, 2020-11-09:
Maybe the problem is related to i2c Bus Speed. U8g2 will probably enable fast i2c. If the Sensor doesn't support fast if and the sensor lib doesn't reset speed then the sensor will not work. 
Try to reduce bus clock speed for u8g2 to 100000 (setBusClock).

v1993, 2020-11-09:
If reducing speed doesn't help (or makes display unacceptably slow), using SW I2C for sensor is probably the most sensible solution. Now that I've looked it up, it's probably SPI driver on ESP-IDF that is much better in multidevice than one from Arduino wrapper, not I2C (if it's clock issue as suspected).

evanroode, 2020-11-10:
I was running in the same problem.

afaict HW I2c wont work on Arduino Uno even when not using any other i2c devices. Also tried an M0 board, and HW-I2C works at high-speed

setting clock speed to 100,000makes it work on my Uno.
```
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R2, SCL, SDA);

void setup(void) {
  u8g2.setBusClock(100000);   <---
  u8g2.begin();
}
```
Now I need to try my code with that M0 board instead and add back code for sensors.

Max13, 2020-11-10:
People of good families... Thanks a lot!

closed 2020-11-10 by Max13

bjrjk, 2021-05-13:
Thank you very much!
This issue solved my problem in my hardware design course-work!