olikraus/u8g2 #1331: ST7920 on 2nd ESP32 SPI BUS (HSPI) in HW mode

hilsonp, 2020-11-10:
I'm using:

- [A ESP32 DEVKIT V1 with 30  GPIO pins](https://s3.us-west-2.amazonaws.com/secure.notion-static.com/06b368b8-8982-49ec-a6e7-64a124f7e061/Untitled.png?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAT73L2G45O3KS52Y5%2F20201109%2Fus-west-2%2Fs3%2Faws4_request&X-Amz-Date=20201109T224356Z&X-Amz-Expires=86400&X-Amz-Signature=bc8cd2790db21d542a0d50a74a319f3fdb41a02a593d7b18fb62ba0165a36745&X-Amz-SignedHeaders=host&response-content-disposition=filename%20%3D%22Untitled.png%22)
- [Full Graphic Smart Controler based on ST7920](https://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf)

My goal is to use this screen on the second SPI bus of the ESP32, in HW mode.

The ESP32 DEVKIT V1 with 30  GPIO pins used are the following:
SPI   MOSI    MISO    CLK     CS
VSPI  GPIO 23 GPIO 19 GPIO 18 GPIO 5
HSPI  GPIO 13 GPIO 12 GPIO 14 GPIO 15

To experiment this, I used the GraphicsTest.ino

My setup() is as follow:
`void setup(void) {
  u8g2.setBusClock(600000);
  u8g2.begin();
}`

Here are my tests (Markdown as messed up the comments around clock/data/reset):
// VSPI SW SPI => OK but slow
//U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ 18, /* data=*/ 23, /* CS=*/ 5, /* reset=*/ U8X8_PIN_NONE); // ESP32
// VSPI HW SPI => OK
//U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, /* CS=*/ 5, /* reset=*/ U8X8_PIN_NONE);
// HSPI SW SPI => OK but slow
//U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ 14, /* data=*/ 13, /* CS=*/ 15, /* reset=*/ U8X8_PIN_NONE); // ESP32
// HSPI HW SPI => NOK
U8G2_ST7920_128X64_F_2ND_HW_SPI u8g2(U8G2_R0, /* CS=*/ 15, /* reset=*/ U8X8_PIN_NONE);

As you can see the 3 first combinations are working.
This let me conclude that:

- My screen is working fine once I:
  - Reduced the speed in the setup()
  - I connected the EXP1-10 pin of the screen board to 5V.
- My connections are OK with the HSPI pins as this works in SW_SPI mode with those pins.

Therefore, I'm raising my hand to get some help.
Is there anything I should change in the library sources to have U8G2_ST7920_128X64_F_2ND_HW_SPI working with my ESP32 ?

Notes: I have been reading plenty of tickets/forum pages before submitting this.

olikraus, 2020-11-28:
In general 2ND HW is not supported for ESP (at least this was the case some time back).
You could change the library code instead. Locate the following function in the mentioned file on your local harddisk:
https://github.com/olikraus/u8g2/blob/1b656cbd5d61d5a907a0f9f32e3b48fd5e2fc3cf/cppsrc/U8x8lib.cpp#L862
Then replace SPI by any other object refering to your second HW SPI.

hilsonp, 2020-12-05:
Thank you for the hint. I will give it a try.

closed 2020-12-05 by hilsonp

hilsonp, 2020-12-14:
So yesterday, I succeeded to make my ESP32 (ESP32 DEVKIT V1) drive my ST7920 (RepRap 12864 LCD) in HW SPI mode on the 2nd SPI BUS (HSPI on pins 12 13 14 & 15) following the advise of @cbpdk : https://github.com/olikraus/u8g2/issues/377#issuecomment-602699767

The advantage of this method is that it only touches the u8g2 files and leaves the ESP32 libraries intact.

So to make it as clear as possible for the next one facing the problem, here is exactly what I did:

1. In U8x8lib.h, add this **before** the #ifdef SPI_INTERFACES_COUNT:
```cpp
/* Declare a second SPI in case of ESP32 */
#if defined(ARDUINO_ARCH_ESP32)
#define SPI_INTERFACES_COUNT 2
#endif
```
2. In U8x8lib.cpp, add this **after** the #include <SPI.h>:
```cpp
/* Declare a second SPI in case of ESP32 */
#include "U8x8lib.h"
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif 
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

/* Declare a second SPI in case of ESP32 */
#if defined(ARDUINO_ARCH_ESP32)
#if SPI_INTERFACES_COUNT > 1
SPIClass SPI1(HSPI);
#endif
#endif
```

With those 2 modifications, I could run the u8g2 GraphicsTest.ino a full night without any problem (note: the speed change in the setup() ):
```cpp
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>

/* HSPI: ST7920 LCD driver in HW SPI mode */
/* LCD EXT1-9 : GND */
/* LCD EXT1-10: 5V */
#define HSPI_MOSI_PIN 13 // To EXP1-3 = LCDE  = ST7920 5 RW(SID)
#define HSPI_MISO_PIN 12 // 
#define HSPI_SCK_PIN  14 // To EXP1-5 = LCD4  = ST7920 6 E(SCK) 
#define HSPI_CS_PIN   15 // To EXP1-4 = LCDRS = ST7920 4 RS(CS)

U8G2_ST7920_128X64_F_2ND_HW_SPI u8g2(U8G2_R0, /* CS=*/ HSPI_CS_PIN, /* reset=*/ U8X8_PIN_NONE); //needed before u8g2.begin() : u8g2.setBusClock(600000); 

void setup(void) {
  u8g2.setBusClock(600000); // Needed with the "RepRap 12864 LCD" (ST7920)
  u8g2.begin();
}

void loop(void) {
  // put your code here
  delay(100);
}
```

I hope this helps.

When I have time, I will re-wire my ESP32 to verify that with this "patch" the fist SPI port can still be used and will report it here (so maybe @olikraus can include this in the library).

HauTranCong, 2022-03-24:
Hi hilsonp !
I used Arduino ide to compile it, but not working for me. Are u using esp-idf for implementation?

hilsonp, 2022-03-25:
I finally used another hardware for this project and can't do any tests right now. 

To reply to your question: Everything was made in Arduino IDE.

JW-Control, 2025-01-20:
Hi hilsonp. Could you share your modified library and ESP32 version you are using? I am currently on ESP32 version 3.0.7 in Arduino IDE, and that may prevent your changes from working properly.

hilsonp, 2025-01-21:
Hello,Sorry. That was a long time ago and this project is not used anymore. Best regards,Pierre