olikraus/u8g2 #1334: Help mapping digital lines for JLX256128

morosanl, 2020-11-12:
Hi, I got this LCD screen "JLX256128G-256, 5V parallel input" and I'm having trouble identifying digital lines. From their datasheet:

8     A0(RS)     Register selection signal     H:   data register 0: instruction register (written as "CD" on the IC   data)
9     RST          Reset                                   Low   level reset, after the reset is completed, return to high level, the LCD   module starts to work
10   CS            Chip   Select                        Low   level chip select
19   RD(E)       Enable   signal                     6800   timing: enable signal
20   WR          Read/write                           6800   Timing: H: Read data L: write data
and here is the constructor:
U8G2_ST75256_JLX256128_1_6800 u8g2(U8G2_R0, d0, d1, d2, d3, d4, d5, d6, d7, en, cs, dc, rst);
So far I've identified :
RD(E) -> en, 
CS -> cs, 
??? - > dc
RST -> rst
Which one goes to dc? Am I correct with the other ones?

[image]

vincent113, 2020-11-13:
DC=RS

olikraus, 2020-11-13:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#wiring

morosanl, 2020-11-13:
Many Thanks!!!

closed 2020-11-13 by morosanl