olikraus/u8g2 #1335: SelectionList Example in combination with Rotary Encoder

Driesvdl, 2020-11-13:
Hi,

I'm inspired by the work you have done! However as a newbie I'm struggling to link things together.
I would really like to have a rotary encoder working instead of the buttons that are defined. I have been searching all of over the internet for many hours but can't seem to get any closer to something that is working.
I would really appreciate your help!
The encoder A is Pin 4, encoder B is pin 3 and the encoder button is pin 5.

`/*

  SelectionList.ino

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

*/

/*
   This is an example of basic menu implemented by U8g2lib.
   For a better usage of ram, we use the Buffer Size mode.

   DaveCalaway
 */

#include <Arduino.h>
#include <U8g2lib.h>

// All Boards without Reset of the Display
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);
byte current_selection = 0;

const char *menu1 =
        "NFC\n"
        "RFID\n"
        "Cirrocumulus\n"
        "Cirrostratus";

const char *menu2 =
        "CCC\n"
        "DDD\n"
        "Cirrocumulus\n"
        "Cirrostratus";

//
// This function draw the text
//
void drawText(byte X,byte Y,const char *text){
        u8g2.firstPage();
        do {
                u8g2.drawStr(X,Y,text);
        } while ( u8g2.nextPage() );
        delay(1000);
}

//
// This function draw the menu
//
void select_menu(const char *nameMenu, const char *select_menu){
        current_selection = u8g2.userInterfaceSelectionList(
                nameMenu,
                current_selection,
                select_menu);
        delay(15);         // debounce
}

//
//
//
void draw_NFC(){
        drawText(15,15,"AAAA");
}

//
// RFID menu
//
void draw_RFID(){
        current_selection = 0;
        bool exit = 0;
        while(exit != 1) {
                const char *nameMenu = "RFID Menu";
                select_menu(nameMenu, menu2);

                if ( current_selection == 0 ) {         // home/cancel button
                        exit = 1;
                }
                if (current_selection == 1) {         // NFC
                        draw_NFC();
                }
        }

}

// ====================================================================================
void setup() {
        u8g2.begin(/*Select=*/ 5, /*Right/Next=*/ A1, /*Left/Prev=*/ A2, /*Up=*/ A0, /*Down=*/ A3, /*Home/Cancel=*/ 8);

        // Set font
        u8g2.setFont(u8g2_font_6x12_tr);

        // Welcome Splash Screen
        drawText(15,40,"Welcome");
}

void loop() {
        const char *nameMenu = "Main Menu";
        select_menu(nameMenu, menu1);

        if ( current_selection == 0 ) {         // home/cancel button
                //
        }
        if (current_selection == 1) {         // NFC
                draw_NFC();
        }
        if (current_selection == 2) {         // RFID
                draw_RFID();
        }
}`

v1993, 2020-11-13:
Please use ``` to format multiline code.

Driesvdl, 2020-11-13:
> Please use ``` to format multiline code.

sry, adapted.

olikraus, 2020-11-28:
The selection list is just an example. Adding support for a rotary encoder will be difficult. At least you need a quadrature decoder software and update the selection list.
I suggest to use https://github.com/neu-rah/ArduinoMenu which supports u8g2 and rotary encoder.

closed 2020-12-25 by olikraus

premoswal, 2021-12-05:
Arduino: 1.8.16 (Windows Store 1.8.51.0) (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

rotaryEvent:39:47: error: call of overloaded 'QDecoder(const int&, const int&, bool)' is ambiguous

 QDecoder qdec(ROTARY_PIN_A, ROTARY_PIN_B, true); // rotary part

                                               ^

In file included from C:\Users\premo\AppData\Local\Temp\arduino_modified_sketch_580125\rotaryEvent.ino:28:0:

C:\Users\premo\Documents\Arduino\libraries\QDEC-master\src/qdec.h:263:34: note: candidate: SimpleHacks::QDecoder::QDecoder(uint16_t, uint16_t, boolean)

     SIMPLEHACKS_INLINE_ATTRIBUTE QDecoder::QDecoder(uint16_t pinA, uint16_t pinB, boolean useFullStep) :

                                  ^

C:\Users\premo\Documents\Arduino\libraries\QDEC-master\src/qdec.h:247:34: note: candidate: SimpleHacks::QDecoder::QDecoder(int16_t, int16_t, boolean)

     SIMPLEHACKS_INLINE_ATTRIBUTE QDecoder::QDecoder(int16_t pinA, int16_t pinB, boolean useFullStep) :

                                  ^

C:\Users\premo\AppData\Local\Temp\arduino_modified_sketch_580125\rotaryEvent.ino: In function 'void setup()':

rotaryEvent:47:13: error: 'D21' was not declared in this scope

 #define SDA D21

             ^

C:\Users\premo\AppData\Local\Temp\arduino_modified_sketch_580125\rotaryEvent.ino:155:14: note: in expansion of macro 'SDA'

   Wire.begin(SDA,SCL);

              ^

rotaryEvent:48:13: error: 'D22' was not declared in this scope

 #define SCL D22

             ^

C:\Users\premo\AppData\Local\Temp\arduino_modified_sketch_580125\rotaryEvent.ino:155:18: note: in expansion of macro 'SCL'

   Wire.begin(SDA,SCL);

                  ^

exit status 1

call of overloaded 'QDecoder(const int&, const int&, bool)' is ambiguous

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.