olikraus/u8g2 #1338: u8x8 text flipped after a while

boerti71, 2020-11-18:
Hi
I'm using WEMOS D1 mini pro and an OLED from CN. it's working fine some but after a while the diplayed text is mirrored in the OLED, without doing some selection.
I'm using 
U8X8_SSD1306_128X64_NONAME_SW_I2C OLED(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // OLEDs without Reset of the Display

when I reboot the Wemos, the display is doing fine again for a while.
Any ideas?

Thanks

olikraus, 2020-11-28:
Maybe there is a kind of a power issue, which leads to a bit change in the memory of the display controller. Place a 1uF cap close to the power supply lines of the OLED.

boerti71, 2020-12-02:
will try it. as it seems to be a hardware issue, I close this issue
thanks for your help!

closed 2020-12-02 by boerti71