olikraus/u8g2 #1339: Question about the 'How can I use multiple SPI Displays' FAQ answer

liamlacey, 2020-11-19:
Hi,

In the 'How can I use multiple SPI Displays' FAQ entry it states "Ensure that the buffer is sent after printing with each display as follows..."

Does this logic apply to _all_ types of drawing (e.g. drawStr(), drawFrame() etc..) or just print()?

Also does this logic also apply to C implementations or just Arduino / C++ implementations?

Thanks!

liamlacey, 2020-11-23:
To give more context about my original question and to question a strange behaviour...

I'm currently attempting to get three OLEDs running using an Atmel SAM MCU (and therefore I'm using the C interface of u8g2) using hardware SPI where each OLED uses its own chip select.

All works absolutely fine with the below code (where I'm calling SendBuffer _directly_ after the drawing code for each screen, and where each OLED is of the _same_ type):

```
u8g2_t screen1;
u8g2_t screen2;
u8g2_t screen3;

void init()
{
   u8g2_Setup_ssd1306_128x64_noname_f (&screen1, U8G2_R0, screen_u8x8_byte_4wire_spi_screen1, screen_u8g2_gpio_and_delay_screen1);
   u8g2_InitDisplay(&screen1);
   u8g2_SetPowerSave(&screen1, 0);

   u8g2_Setup_ssd1306_128x64_noname_f (&screen2, U8G2_R0, screen_u8x8_byte_4wire_spi_screen2, screen_u8g2_gpio_and_delay_screen2);
   u8g2_InitDisplay(&screen2);
   u8g2_SetPowerSave(&screen2, 0);

   u8g2_Setup_ssd1306_128x64_noname_f (&screen3, U8G2_R0, screen_u8x8_byte_4wire_spi_screen3, screen_u8g2_gpio_and_delay_screen3);
   u8g2_InitDisplay(&screen3);
   u8g2_SetPowerSave(&screen3, 0);
}

void update()
{
   u8g2_ClearBuffer(&screen1);
   u8g2_SetDrawColor(&screen1, 1);
   u8g2_SetFont(&screen1, u8g2_font_logisoso16_tf);
   u8g2_DrawStr(&screen1, 1, 18, "SCREEN 1");
   u8g2_SendBuffer(&screen1);

   u8g2_ClearBuffer(&screen2);
   u8g2_SetDrawColor(&screen2, 1);
   u8g2_SetFont(&screen2, u8g2_font_logisoso16_tf);
   u8g2_DrawStr(&screen2, 1, 18, "SCREEN 2");
   u8g2_SendBuffer(&screen2);

   u8g2_ClearBuffer(&screen3);
   u8g2_SetDrawColor(&screen3, 1);
   u8g2_SetFont(&screen3, u8g2_font_logisoso16_tf);
   u8g2_DrawStr(&screen3, 1, 18, "SCREEN 3");
   u8g2_SendBuffer(&screen3);
}
```

However if I change the ordering of the code so that _all_ drawing is done first and the SendBuffer calls are done _together_ at a later time (which I'd ideally like to do), like as follows, then _all_ screens display the buffer for screen3 (buffers for screen1 and screen2 appear to be ignored):

```
void update()
{
   u8g2_ClearBuffer(&screen1);
   u8g2_SetDrawColor(&screen1, 1);
   u8g2_SetFont(&screen1, u8g2_font_logisoso16_tf);
   u8g2_DrawStr(&screen1, 1, 18, "SCREEN 1");
  
   u8g2_ClearBuffer(&screen2);
   u8g2_SetDrawColor(&screen2, 1);
   u8g2_SetFont(&screen2, u8g2_font_logisoso16_tf);
   u8g2_DrawStr(&screen2, 1, 18, "SCREEN 2");

   u8g2_ClearBuffer(&screen3);
   u8g2_SetDrawColor(&screen3, 1);
   u8g2_SetFont(&screen3, u8g2_font_logisoso16_tf);
   u8g2_DrawStr(&screen3, 1, 18, "SCREEN 3");
   
   ...

   u8g2_SendBuffer(&screen1);
   u8g2_SendBuffer(&screen2);
   u8g2_SendBuffer(&screen3);
}
```

This first led me to believe I couldn't order the code in this way based on the FAQ answer I originally highlighted.

However, if I instead use three _different_ types of OLEDs (e.g. an SSD1305, an SSD1306 and an SSD1317) and use the relevant `u8g2_Setup_` functions then the above problem goes away - all screens draw their own buffers. Though I did find in this that SSD1306 and SSD1309 screens appear as the same type in relation to this issue.

Is this the expected behaviour?
If not, is this a bug or what am I doing something wrong?

Many thanks.

olikraus, 2020-11-23:
> Is this the expected behaviour?

Well, it is like this: U8g2 was designed for low memory footprint and to work without memory allocation.
There is a static allocation mechanism in u8g2 which will reserve memory for a device. If such memory has been allocated before, then the next device will use the same memory, otherwise it will reserver a new memory.

First device allocates memory for a 128x64 memory. Second device is a 128x128 device and will allocate a new block with 128x128 pixel. Third device again requires 128x64 pixel and will share the memory with the first device (because it is exactly the same size).

This means for your second example:
```
   u8g2_ClearBuffer(&screen1);
   u8g2_SetDrawColor(&screen1, 1);
   u8g2_SetFont(&screen1, u8g2_font_logisoso16_tf);
   u8g2_DrawStr(&screen1, 1, 18, "SCREEN 1");
  
   u8g2_ClearBuffer(&screen2); /* this will overwrite screen1 of both frame buffers have same size */
```

liamlacey, 2020-11-23:
Thanks for the response - that all makes sense.

Is there an easy way to enable / force individual memory allocation for each device, if a low memory footprint isn't a requirement / issue?

olikraus, 2020-11-23:
> Is there an easy way to enable / force individual memory allocation for each device, if a low memory footprint isn't a requirement / issue?

No, it is kind of hard coded into u8g2. The problem here is the missing dynamic memory allocation: Everything is predefined via c generated source code. 

There was once a pull request, which has implemented dynamic memory allocation. Maybe this helps: If U8G2_USE_DYNAMIC_ALLOC is defined, then maybe dynamic memory is used.

But I have not tested this...

liamlacey, 2020-11-23:
Many thanks for this.

I was not aware of this mode - this appears to have solved the problem for me.

Thanks!

closed 2020-11-24 by liamlacey