olikraus/u8g2 #1340: Print a variable without leading zeros

Betacarotin, 2020-11-20:
Hi! 
I am trying to solve this for quite some time and don´t know how to tackle a obvious problem:

I would like to display a variable that can be 0-100, 

I am using the u8x8 on a Arduino Nano with an oled display:
U8X8_SH1106_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);
    u8x8.begin();
    u8x8.setPowerSave(0);
    display();

And in my function:
  u8x8.setFont(u8x8_font_amstrad_cpc_extended_r);
  u8x8.setCursor(0, 0);
   u8x8.drawString(0,20, String(pwmd).c_str()); 

Alternative I can use 
  u8x8.print(pwmd);

BUT: When I go from 10 to 9 it will display a 90 

I dont want to have 09, i would like to display 9 only. 

How can I achieve this?

Betacarotin, 2020-11-20:
u8x8.print("%i",pwmd); 
leaves me with an error: no matching function

AnHardt, 2020-11-20:
You are not printing "90" but "9" at the same space where the "10" was written before without erasing the space. So the "0" is still visible.
A 
```cpp
u8x8.drawString(0,20, "   "); // 3 spacers to cover 100
```
before the 
```cpp
u8x8.drawString(0,20, String(pwmd).c_str());
```
should help.
If the font is not monospace, but proportional, you may need more spaces.

Betacarotin, 2020-11-20:
Yes, that works, kind of.
BUT: it still flickers with that noticeable

And can I have all the digits on the right side instead of the left side:
It is:
9
10
100

I want 
_ _9
 _10
100

AnHardt, 2020-11-20:
Just do it.
```
if (number < 100) print(" ") ;
if (number < 10) print(" ") ;
print(number);
```

Betacarotin, 2020-11-20:
Yeah, no- as said it blinks/flickers the output!

I do not understand, my system supports sprintf? It is Arduino IDE..How shal I do that? 
All the syntaxes I tried left me with not compiling code 

like:
  u8x8.drawString(0,20, sprintf("%d",pwmd)); 

HELP please!

Betacarotin, 2020-11-20:
u8x8.drawString(0,20, sprintf("%3d",pwmd));
compiles but do not output anything

AnHardt, 2020-11-20:
http://www.cplusplus.com/reference/cstdio/sprintf/

Betacarotin, 2020-11-20:
How is that supposed to work? When I copy the example given and try to output it it does not compile
  char buffer [50];
  int n, a=5, b=3;
  n=sprintf (buffer, "%d plus %d is %d", a, b, a+b);
  u8x8.printf ("[%s] is a string %d chars long\n",buffer,n);

neither does:
  u8x8.print ("[%s] is a string %d chars long\n",buffer,n);

Betacarotin, 2020-11-20:
Nor does:
`  char buffer [50];
  int n, a=5, b=3;
  n=sprintf (buffer, "%d plus %d is %d", a, b, a+b);

  u8x8.drawString(0,20, n);`

work. Is theer nobody who could provide me some advice please?

Betacarotin, 2020-11-20:
I found this one here:
`  u8x8.print(u8x8_u8toa(pwmd,3));`

from here:
https://github.com/olikraus/u8g2/issues/594

It half- solves the problem- how do I get rid of the leading zeros? I would like to have just blank spaces.

olikraus, 2020-11-28:
I think you could also extend the previous example:
``` C++
if (number < 100) print(" ");
if (number < 10) print(" ");
print(number);
print("  ");  // clear previous content after the number
```
regarding u8toa, you could remove the leading zeros:
``` C++
const char *s = u8x8_u8toa(pwmd,3);
for(;;) {
  if ( s[1] == '\0' ) break;
  if ( s[0] >= '1' ) break;
  s++;
}
u8x8.print(s);
```
(code not tested)

closed 2020-12-25 by olikraus