olikraus/u8g2 #1342: Failure with u8g2 usage on Arduino nano BLE sense with LD7032

redflowIT, 2020-11-22:
HI all
using u8g2 on an Arduino NANO BLE Sense with an LD7032 Display (FUTABA) I have strange behaviour: graphics and test start in the middle of the display with a part above the centre and a part belove.
[https://www.dropbox.com/s/qi79ia6l82o3o4a/IMG-0373.JPG?dl=0](url)
[https://www.dropbox.com/s/e081qs70gkcl1oe/IMG-0374.JPG?dl=0](url)
[https://www.dropbox.com/s/ibx5xpp5eg16epl/IMG-0375.JPG?dl=0](url)
These photos are the result of the example full_buffer-GraphicsTest but I've tried almost everything with same results: HW or SW connection, Full-Buffer or Paged....everything with same results.

The display was changed because the one we have bought with PCB had broken flat.

Any suggestion?

Thank you

redflowIT, 2020-11-22:
I'm using this Display:
https://www.mouser.it/datasheet/2/921/Futaba-04-14-2020-ELW0501AB_H_SHORT-1830787.pdf

with this constructor:
`U8G2_LD7032_60X32_F_4W_SW_SPI u8g2(U8G2_R0, /*clock=*/ 11, /*data=*/ 12, /*cs=*/ 9, /*dc=*/ 10, /*reset=*/ 8);`

redflowIT, 2020-11-23:
I've downloaded the complete datasheet hoping it could be useful:
https://www.dropbox.com/s/k2kfqs8mrfifyhf/ELW0501ABR_B_03.pdf?dl=0

olikraus, 2020-11-28:
Your display does not fully fit to the current driver, I mean the existing u8g2 constructor has been written for a different display.

try to adjust the y start position, use some other value then the current 0x00:
https://github.com/olikraus/u8g2/blob/1b656cbd5d61d5a907a0f9f32e3b48fd5e2fc3cf/csrc/u8x8_d_ld7032_60x32.c#L88

You need to locate the file u8x8_d_ld7032_60x32.c on your harddisk and update the file.

olikraus, 2020-12-25:
No further feedback, closing...

closed 2020-12-25 by olikraus