olikraus/u8g2 #1343: Nodemcu and ST7920

marcoayamada, 2020-11-22:
Hi everyone!

I'm struggling with Nodemcu (esp8266) and ST7920 display. A simple clock test program doesn't works fine.

Some pixels in display are mess or/and weird characters are show:

[image]

My code:

```
#include <Arduino.h>
#include <U8g2lib.h>

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R2, 14, 13, 15); 

void setup(void) {
  
  u8g2.begin(); 
}

uint8_t m = 24;

void loop(void) {
  char m_str[3];
  strcpy(m_str, u8x8_u8toa(m, 2));    /* convert m to a string with two digits */
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_logisoso62_tn);
    u8g2.drawStr(0,63,"9");
    u8g2.drawStr(33,63,":");
    u8g2.drawStr(50,63,m_str);
  } while ( u8g2.nextPage() );
  delay(2000);
  m++;
  if ( m == 60 )
    m = 0;
}
```
My breadboard:

[image]

Could anyone help me with this?

olikraus, 2020-11-24:
Looks like a timing or signal quality issue. In fact the ST7920 is the most sensitive display regarding this. You may also shorten your wires and/or use different holes in your protoboard.
You can also try to reduce the bus transfer speed with the SetBusClock command.

olikraus, 2020-11-28:
any success?

marcoayamada, 2020-11-29:
I didn't test modifying the circuit yet. But I set clock to a slow speed and the result was better. I think if I add a capacitor in a circuit this issue will be solved. I will try this week.

Bruno999B, 2020-12-05:
I have the same problem with :
U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R2, D8);

but work perfectly with :
U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R2, 14, 13, 15, U8X8_PIN_NONE);

Bruno999B, 2020-12-05:
Apparently the addition of a Sanyo 3300uf 6.3v capacitor on the esp8266's VIN solved the problem, works well in hardware SPI at 1mhz.

olikraus, 2020-12-25:
hmmm, I guess it's time to close this issue...

closed 2020-12-25 by olikraus