olikraus/u8g2 #1344: STM32G0 and u8x8 - nothing happens

SolarNinjaTurtle, 2020-11-27:
Hi,
im trying to implement your u8x8 library to my STM32G0 project using HAL and FreeRTOS.
I have an 128x32 OLED 0.91" NoName with SSD1306 from China. Connection is okay, worked great with another library.
Using hardware i2c with 400khz.

I implemented the needed functions like this:

> 
> uint8_t u8x8_byte_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
> 	/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
> 	static uint8_t buffer[128];
> 	static uint8_t buf_idx;
> 	uint8_t *data;
> 
> 	switch (msg) {
> 	case U8X8_MSG_BYTE_INIT: {
> 		/* add your custom code to init i2c subsystem */
> 		MX_I2C2_Init();
> 	}
> 		break;
> 
> 	case U8X8_MSG_BYTE_START_TRANSFER: {
> 		buf_idx = 0;
> 	}
> 		break;
> 
> 	case U8X8_MSG_BYTE_SEND: {
> 		data = (uint8_t*) arg_ptr;
> 
> 		while (arg_int > 0) {
> 			buffer[buf_idx++] = *data;
> 			data++;
> 			arg_int--;
> 		}
> 	}
> 		break;
> 
> 	case U8X8_MSG_BYTE_END_TRANSFER: {
> 		if (HAL_I2C_Master_Transmit(&hi2c2, 0x78, buffer, buf_idx, 1000) != HAL_OK)
> 			return 0;
> 	}
> 		break;
> 
> 	case U8X8_MSG_BYTE_SET_DC:
> 		break;
> 
> 	default:
> 		return 0;
> 	}
> 
> 	return 1;
> }
> 
> uint8_t u8x8_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
> 	switch (msg) {
> 	case U8X8_MSG_DELAY_100NANO: // delay arg_int * 100 nano seconds
> 		__NOP();
> 		break;
> 	case U8X8_MSG_DELAY_10MICRO: // delay arg_int * 10 micro seconds
> 		for (uint16_t n = 0; n < 320; n++) {
> 			__NOP();
> 		}
> 		break;
> 	case U8X8_MSG_DELAY_MILLI: // delay arg_int * 1 milli second
> 		HAL_Delay(1);
> 		break;
> 	case U8X8_MSG_DELAY_I2C: // arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
> 		delay_us(5);
> 		break;                    // arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us
> 	case U8X8_MSG_GPIO_I2C_CLOCK: // arg_int=0: Output low at I2C clock pin
> 		break;                    // arg_int=1: Input dir with pullup high for I2C clock pin
> 	case U8X8_MSG_GPIO_I2C_DATA:  // arg_int=0: Output low at I2C data pin
> 		break;                    // arg_int=1: Input dir with pullup high for I2C data pin
> 	case U8X8_MSG_GPIO_MENU_SELECT:
> 		u8x8_SetGPIOResult(u8x8, /* get menu select pin state */0);
> 		break;
> 	case U8X8_MSG_GPIO_MENU_NEXT:
> 		u8x8_SetGPIOResult(u8x8, /* get menu next pin state */0);
> 		break;
> 	case U8X8_MSG_GPIO_MENU_PREV:
> 		u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */0);
> 		break;
> 	case U8X8_MSG_GPIO_MENU_HOME:
> 		u8x8_SetGPIOResult(u8x8, /* get menu home pin state */0);
> 		break;
> 	default:
> 		u8x8_SetGPIOResult(u8x8, 1); // default return value
> 		break;
> 	}
> 	return 1;
> }
> 
> void delay_us(uint32_t time) {
> 	uint32_t i = 8 * time;
> 	while (i--)
> 		;
> }

Setup the u8x8 like this:

> u8x8_Setup(&u8x8, u8x8_d_ssd1306_128x32_winstar, u8x8_cad_ssd13xx_fast_i2c, u8x8_byte_hw_i2c, u8x8_gpio_and_delay);
> u8x8_InitDisplay(&u8x8);
> u8x8_ClearDisplay(&u8x8);
> u8x8_SetFont(&u8x8, u8x8_font_chroma48medium8_r);
> u8x8_DrawString(&u8x8, 10, 10, "Hello World!");
> 

After debugging i can say that the controller stuck in the function u8x8_InitDisplay(&u8x8);, when im ignore the line the controller does not stuck, but the display shows nothing.

Do you have an idea what my problem could be?

Thanks,
Paul

SolarNinjaTurtle, 2020-11-27:
It stucked on HAL_Delay(1);

Should be osDelay(1), now works rudimentary. Cant really read the text on the display.

SolarNinjaTurtle, 2020-11-27:
It works now with "u8x8_d_ssd1306_128x32_univision", but i dont have an univision. 
Its an NoName 128x32, but i cant find it in your list.

The Y Cursor dont work :-(

olikraus, 2020-11-28:
"univision" is just a placeholder. The init code often fits to other displays as well, so using the univision code should be ok.

closed 2020-12-25 by olikraus