olikraus/u8g2 #1345: Missing pixels when positioned at 0 or 1 in  aline of text (SH1106 OLED)

NormanDunbar, 2020-11-28:
When I execute the Hello World example (below) on an Arduino Uno and on a Wemos D1 Mini, the text output is incorrect.

* With the line `u8g2.drawStr(0,10,"Hello World!");` the first two pixels in each column of the leading 'H' are missing from the left of the OLED, but are displayed at the far right of the OLED. 
* With the line `u8g2.drawStr(1,20,"Hello World!");` the first one pixel in each column of the leading 'H' is missing from the left of the OLED, but two columns of pixels like `|` are displayed at the far right of the OLED. 
* With the line `u8g2.drawStr(2,30,"Hello World!");` the text is correctly displayed, but two columns of pixels which resemble the right two columns of pixels from the leading "H" are displayed at the far right of the OLED. 
* With the line `u8g2.drawStr(4,30,"Hello World!");` the text is correctly displayed, with no spurious pixels at the far right of the line.

Calling `clearBuffer(); sendBuffer()` clears the text, but leaves the two columns of spurious pixels at the far right untouched.

The data sheet for the SH1106 designates it as a 132 by 64 display -- I wonder if that's affecting things?

The constructor used in the sketch is `U8G2_SSD1306_128X64_NONAME_F_HW_I2C oled(U8G2_R0, U8X8_PIN_NONE);` The problem exists similarly on the three fonts I have attempted to use:

* u8g2_font_ncenB08_tr - shows the missing pixels plus spurious pixels to the right.
* u8g2_font_6x10_tf - shows the missing pixels plus similar spurious pixels to the right.
* u8g2_font_6x12_tf - loses pixels on the left, but no spurious pixels appear on the far right of each line. 

If I offset the text coordinates to 2, instead of 0, I see the whole text. Clearing the buffer doesn't remove the pixels at the far right of the OLED, only powering off and back on clears those.

Example code to demonstrate the problem. 

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ 16, /* data=*/ 17);   // ESP32 Thing, HW I2C with pin remapping

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();

  // Pick a font, any font...
  //u8g2.setFont(u8g2_font_6x10_tf);
  //u8g2.setFont(u8g2_font_6x12_tf);
  u8g2.setFont(u8g2_font_ncenB08_tr);

  // This loses two columns of pixels from the left side, 
  // but displays them at the far right where they are
  // never erased.
  u8g2.drawStr(0,10,"Hello World!");
  
  // This is broken too. Also displays pixels on the 
  // far right.
  u8g2.drawStr(1,20,"Hello World!");

  // This correctly displays the text, but also shows spurious
  // pixels on the right side of the OLED.
  u8g2.drawStr(2,30,"Hello World!");

  // This "just" works!
  u8g2.drawStr(4,40,"Hello World!");
  
  u8g2.sendBuffer();

  // Delay 5 seconds to view results.
  delay(5000);  

  // This clears the text, but not the spurious pixels.
  u8g2.clearBuffer();
  u8g2.sendBuffer();

  // Wait 2 seconds to view the problem.
  delay(2000);  
}
```

Regards,
Norm.

olikraus, 2020-11-28:
Probably you should use the SH1106 constructor instead of the SSD1306 constructor.

https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1106-128x64_noname

NormanDunbar, 2020-11-28:
Ah! Oops! Silly me. 😢 

In my defence, I was following instructions in the manual that came with my OLED. I was wondering about the numbers and did check but the docs are obviously wrong.

Thank you for the quick response, and my apologies for wasting your time. Sorry. I'll raise the matter with the supplier of the OLED. Thanks again.

Stay safe.

Cheers,
Norm.

closed 2020-11-28 by NormanDunbar