olikraus/u8g2 #1346: Missing Fonts?

trailsurfer604, 2020-11-29:
I apologize in advance, this is probably an extremely stupid question, but for the life of me I can't figure out how to install / ensure that I can use all fonts from https://github.com/olikraus/u8g2/wiki/fntlistall.

I am using Adafruit GFX for ST7735 with U8g2_for_Adafruit_GFX.h.  I have downloaded the latest master.

At https://github.com/olikraus/u8g2/wiki/fntlistall, under 15 point fonts, there is a font called "u8g2_font_VCR_OSD_tr"

However, it is not listed in U8g2_for_Adafruit_GFX.h, and neither are many other fonts listed in the font install.

How do I get these fonts?

olikraus, 2020-11-29:
I probably need to update U8g2_for_Adafruit_GFX.h

Alternative solution: you could copy the single  font array to your .ino file.

trailsurfer604, 2020-11-30:
So I finally got a clue, used u8g2_fonts.c from the latest u8g2 release, and followed 3 of your 5 steps for updating the font list:

    1. Copy font file u8g2_fonts.c to "src" folder
    2. Change the include file in u8g2_fonts.c from "u8g2.h" to "u8g2_fonts.h"
    3. Update the list of fonts in "U8g2_for_Adafruit_GFX.h"
    4. Update keywords.txt
    5. Update version in library.properties

I don't quite know what to do for steps 4 and 5, but I don't think that's what's causing the new compile error:

`libraries\U8g2_for_Adafruit_GFX\u8g2_fonts.c.o (symbol from plugin): In function `u8g2_font_px437wyse700b_mn':
(.text+0x0): multiple definition of `u8g2_font_px437wyse700b_mn'
libraries\U8g2_for_Adafruit_GFX\u8g2_fonts - Copy.c.o (symbol from plugin):(.text+0x0): first defined here
`

This is the last font defined in u8g2_fonts.c and "U8g2_for_Adafruit_GFX.h.  I checked both files, the font definition is identical to what it was before I updated them. 

What stupid thing am I missing now?

trailsurfer604, 2020-11-30:
Thank you, I finally got a clue and figured out how to read your instructions for updating the font base

I.e. I downloaded the latest U8g2 library and copied the font pack, with necessary modifications, to U8g2_for_Adafruit_GFX.

Works now.

closed 2020-11-30 by trailsurfer604