olikraus/u8g2 #1349: SH1106 display area is 2 pixels off the actual screen area

ValChinnici, 2020-12-01:
I have a 1.3" Oled display, and I am trying to test it with an Arduino Uno.

The library works, as I am able to display the examples included in the library... However I realised the display area has shifted 2 pixels on the left. It happens to be a very common issue people have been experiencing, but I haven't find a solution for it yet. I have tried many examples and looked for solutions online, nothing seems to work. Below a picture of the oled screen with the highlited area that has shifted. 

[image]

olikraus, 2020-12-06:
Will the ssd1306 driver work?

VKin-Arduino, 2020-12-06:
> Will the ssd1306 driver work?

I tried, if I use that driver the oled doesn't light up :(

olikraus, 2020-12-06:
hmm, the ssd1306 should work.

You could try to set the x offset to 0:
https://github.com/olikraus/u8g2/blob/1b656cbd5d61d5a907a0f9f32e3b48fd5e2fc3cf/csrc/u8x8_d_ssd1306_128x64_noname.c#L394-L395
However, this is the main difference between sh1106 and ssd1306

closed 2020-12-25 by olikraus