olikraus/u8g2 #1350: Epson E-1330BA LCD Screen

amd989, 2020-12-01:
Hi, 

I came across this large LCD board with ID PJUP397ZB, it's being driven by an Epson E-1330BA, I can only assume this chip is analog to a SED1330FBA or similar, as the pins connected to the main connector follow the same pattern as the [SED1330F datasheet](https://www.lcd-module.de/eng/pdf/zubehoer/sed1330.pdf) pinout. There is almost no information I could find about this LCD online. It came from an old [KX-WL50 ](http://salestores.com/panasonickxw.html) word processor. And is apparently 80x14 if that ad is to be believed.

Anyways, I'm having trouble getting it to work, after finding all it's relevant pins, it seems to show some signs of life, but I can't get anything to show up other than some lines. 

The board seems to be hardwired for 6800. as SEL1 and SEL2 are connected to VCC and GND respectively matching this table:
[image]

Here what I did:

| Arduino | Connection  |
|---------|-------------|
| 0-7     | d0-d7       |
| 8       | Enable      |
| 9       | Chip Select |
| 10      | A0 / DC       |
| 11      | Reset       |
| GND     | R/W         |

Having flashed the HelloWorld example. After connecting power I see some lines in the screen (some lines may be faulty), then it gets cleared (by the reset sequence, I suppose) then nothing.

I have used the following constructors
`U8G2_SED1330_240X128_2_6800`
`U8G2_RA8835_320X240_1_6800`
`U8G2_RA8835_NHD_240X128_1_6800`

I have turned on/off 16 bit mode but seemingly makes no difference.

There is only one extra pin in the connector that's left. But is not connected to the 1330BA but is rather connected to the "**DF**" pin in all the "80-DOT LCD segment drivers" MSM5298A and MSM5299A in the board.

Which is described as

> **DF**
> Synchronous signal input pin for alternate signal for LCD driving.
> 

And follows this table according to the datasheet
[image]

Connecting this Pin to GND or otherwise touching it, makes some faint but visible lines to appear in the screen. 

I'm using Arduino UNO.

Any help would be appreciated.

amd989, 2020-12-01:
If I leave it on for a while it eventually flashes into this

[image]

olikraus, 2020-12-25:
I think you already tested the correct constructors. I can not say much about wiring. 
One more thing to consider is the LCD voltage: It is either a negative or positive voltage maybe of -15 or 15V. If this voltage is not generated internally, then it needs to be provided via one of the pins. Did you measure this voltage somewhere? If this voltage is not available, then nothing will be visible.

closed 2021-05-09 by olikraus