olikraus/u8g2 #1351: Support for SSD1320

ddlqn, 2020-12-03:
I was hoping to use u8g2 for [this 160x32 display](https://www.sparkfun.com/products/14606), but it uses the SSD1320 controller which is currently not supported. I have used the library with a different display before and think it is great. So I was wondering whether there is any chance you could add support for the SSD1320 controller (I realise the display is 4bit greyscale, but I'd be ok using it as a monochrome display for now).

Thanks!

olikraus, 2020-12-06:
Did you test any of the existing SSD132x drivers?

ddlqn, 2020-12-07:
I had a look through the command tables and there are some differences. I started creating my own u8x8_d_ssd1320.c. What other modules need to be changed/added? Happy to share if I get it to work.

ddlqn, 2020-12-15:
Update: I got it to work, but for some reason only in U8G2_MIRROR mode. Am I missing something? Would you like me to create a pull request?

olikraus, 2020-12-21:
I can create a new constructor. What exactly did you add? Can you paste u8x8_d_ssd1320.c here?
the memory.c and setup.c files are automatically generated.

olikraus, 2020-12-25:
How shell we continue here?

buildxyz-git, 2021-02-05:
Hi @olikraus, your amazing library has saved me tons of time on my journey to [convert a pinball machine into a coffee table](https://www.buildxyz.xyz/hacking-a-pinball-machine-into-a-coffee-table/), thank you so much!

I'm currently experimenting with a large display which has dual SSD1320Z controllers. I sparked up a discussion [here](https://github.com/olikraus/u8g2/discussions/1396) with some initial information and observations. I'm wondering if you have any insight/recommendations/things to try?

olikraus, 2021-02-07:
Hi @buildxyz-git 
I have seen the discussion, but I think the creation of the SSD1320 would be a precondition. Once this is available, you can just create two u8g2 objects with two different CS lines.

olikraus, 2021-02-07:
I would order such a display to write and test a driver, but it looks like that the available displays are little bit expensive for a private spare time project like u8g2.

buildxyz-git, 2021-02-07:
I understand, it is an expensive display. It makes good sense the create a working SSD1320 driver first. 

If that was completed, I would have to treat the display as two separate displays in my code. An example of this is printing a string across the entire display which I plan to do to represent the current players score.

Can you see a simple way to wrap all the great features already incorporated into u8g2 (drawXBM etc) so they could still be called without splitting images or doing additional math before each print operation?

olikraus, 2021-02-07:
Using.two displays as one will be simple. Just draw everything on both displays with suitable offset. U8g2 will do the clipping. I worry more about the setup.

buildxyz-git, 2021-02-07:
I can supply a display for you to work with if this is something you are interested in implementing. The library is really awesome and I'm happy to support it :)

The manufacturer has supplied me with great set of working documentation including the SSD1320 data sheet and some example code for a STM32 in a Keil project.

Send [me](https://github.com/buildxyz-git) an email if you want to work on this together!

buildxyz-git, 2021-02-08:
I’m also able to provide the data sheet and test any SSD1320 constructors and report back here with results!

buildxyz-git, 2021-02-09:
[SSD1320.pdf](https://github.com/olikraus/u8g2/files/5947679/SSD1320.pdf)

buildxyz-git, 2021-02-09:
[SUR383S1000WG01_Module.pdf](https://github.com/olikraus/u8g2/files/5948977/SUR383S1000WG01_Module.pdf)

olikraus, 2021-02-09:
Thanks for all the support and documents. I will implement a new constructor based on your docs. Maybe we both can make that thing working if you do some testing.

buildxyz-git, 2021-02-09:
That would be really great @olikraus! I read through the data sheet and started a cross referencing other SSD13XX constructors but I think I need a better understand of how the library populates the display RAM. This constructor at turns the display on and prints garbage on half the screen.
`/*

  u8x8_d_ssd1320.c

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  
  
*/

#include "u8x8.h"

/* ER OLED */
static const uint8_t u8x8_d_ssd1320_xyz_160x132_init_seq[] = {
    
  U8X8_START_TRANSFER(),    /* enable chip, delay is part of the transfer start */

  U8X8_C(0x0ae),		        /* display off */
  U8X8_CA(0x0fd, 0x012),		/* unlock (not required, this is default by reset) */

  U8X8_CA(0x20, 0x00),    /* set addressing mode */
  U8X8_CA(0x25, 0x00),      /* set portrait addressing mode - normal addressing mode */
  U8X8_CA(0x81, 0x6b),    /* set contrast */

  // U8X8_C(0xa0),             /* disable column remap */
  U8X8_C(0xa1),             /* enable column remap */
  // U8X8_CA(0x20, 0x00),      /* horizontal address increment */
  U8X8_C(0xc8),             /* enable COM Remap */

  // U8X8_C(0xa1),            /* set seg remap CS2 */
  U8X8_CA(0x0a2, 0x000),    /* set Display line start */
  U8X8_CA(0x0a8, 0x083),		/* set mux ratio: 1/132 duty */ 
  U8X8_CA(0x0ad, 0x010),    /* select internal or external IREF */
  U8X8_CA(0x0bc, 0x01e),    /* precharge voltage */
  U8X8_C(0x0bf),            /* linear LUT */
  U8X8_C(0x0c8),            /* set COM output scan direction CS1 */
  // U8X8_C(0x0c0),            /* set COM output scan direction CS2 */
  U8X8_CA(0x0d3, 0x00e),    /* set display offset CS1 */
  // U8X8_CA(0x0d3, 0x092),    /* set display offset CS2 */
  U8X8_CA(0x0d5, 0x0c2),    /* set clock divide ratio / osc freq - 8hHz*/       
  U8X8_CA(0x0d9, 0x072),    /* Set pre charge period */
  
  U8X8_CA(0x0da, 0x032),    /* SEG pins hardware configuration */
  // U8X8_CA(0x0da, 0x012),
[... 133 lines cut ...]
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 15,
  /* write_pulse_width_ns = */ 150,	/* SSD1306: cycle time is 300ns, so use 300/2 = 150 */
  /* tile_width = */ 20,
  /* tile_hight = */ 7,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 160,
  /* pixel_height = */ 132
};

uint8_t u8x8_d_ssd1320_xyz_160x132(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    if ( msg == U8X8_MSG_DISPLAY_SETUP_MEMORY )
    {
      u8x8_d_helper_display_setup_memory(u8x8, &u8x8_d_ssd1320_xyz_160x132_display_info);
      return 1;
    }
    return u8x8_d_ssd1320_xyz_160x132_generic(u8x8, msg, arg_int, arg_ptr);
}`

using the above constructor and running the GraphicsTest.ino sketch produces the following output:

https://user-images.githubusercontent.com/16586332/107393685-ed6b8600-6aaf-11eb-86fb-857f2739bec2.MOV

ddlqn, 2021-02-09:
Apologies for not following up on this earlier. If it helps, this is the constructor I created for my project. Maybe it can serve as a starting point for one that is more general. As mentioned in one of my earlier comments, one of the issues is that it only works in U8G2_MIRROR mode.

```
/*

  u8x8_d_ssd1320.c
  
  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

  SSD1320: 
    160 x 160 dot matrix
    16 gray scale
  
  Adapted from u8x8_d_ssd1322.c with the command set of the SSD1320 controller
  "official" procedure is described here: https://github.com/olikraus/u8g2/wiki/internal
  
*/
#include "u8x8.h"

static const uint8_t u8x8_d_ssd1320_powersave0_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x0af),		                /* ssd1320: display on */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

static const uint8_t u8x8_d_ssd1320_powersave1_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x0ae),		                /* ssd1320: display off */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

/*
  input:
    one tile (8 Bytes; 1 byte per column)
  output:
    Tile for SSD1320 (32 Bytes)

  The origin of the display seems to be in the upper right-hand corner. Therefore
  compared to SSD1322, the order inside each byte is swapped.
*/

[... 184 lines cut ...]
      break;
    case U8X8_MSG_DISPLAY_INIT:
      u8x8_d_helper_display_init(u8x8);
      u8x8_cad_SendSequence(u8x8, u8x8_d_ssd1320_160x32_init_seq);
      break;
    case U8X8_MSG_DISPLAY_SET_FLIP_MODE:
      if ( arg_int == 0 )
      {
	u8x8_cad_SendSequence(u8x8, u8x8_d_ssd1320_160x32_flip0_seq);
	u8x8->x_offset = u8x8->display_info->default_x_offset;
      }
      else
      {
	u8x8_cad_SendSequence(u8x8, u8x8_d_ssd1320_160x32_flip1_seq);
	u8x8->x_offset = u8x8->display_info->flipmode_x_offset;
      }
      break;
    
    default:
      return u8x8_d_ssd1320_common(u8x8, msg, arg_int, arg_ptr);
  }
  return 1;
}

```

buildxyz-git, 2021-02-10:
Well that was the best thing that happened today @ddlqn. The constructor works very well! I'm still unclear as to how initialize each controller to get a seamless image.

I was able to fix the mirroring issue by changing this:
`U8X8_C(0xa0),	                /* Set Segment Re-Map: column address 0 mapped to SEG0 */  `

To this:
`U8X8_C(0xa1),	                /* Set Segment Re-Map: column address 0 mapped to SEG0 */  `

You can see from the pictures that my resolution is off and using CS1 results in upside image on the left half of the screen, but progress was made!

CS1:
[image]

CS2:
[image]

if did update the following:
```
  /* tile_width = */ 20,		/* 160 pixel, so we require 20 bytes for this */
  /* tile_hight = */ 7,
  /* pixel_width = */ 160,
  /* pixel_height = */ 132
```
I also updated the display offset in the init commands because I saw the manufacture example code did this and it displayed better:
`U8X8_CA(0xd3, 0x93),`

buildxyz-git, 2021-02-10:
I was able to make a little more progress, I believe I will need slightly different setup commands for each side of the display. I'm just trying to get the left side of the display working properly first. the current ssd1320.c is as follows: 
```
/*

  u8x8_d_ssd1320.c
  
  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

  SSD1320: 
    160 x 160 dot matrix
    16 gray scale
  
  Adapted from u8x8_d_ssd1322.c with the command set of the SSD1320 controller
  "official" procedure is described here: https://github.com/olikraus/u8g2/wiki/internal
  
*/
#include "u8x8.h"

static const uint8_t u8x8_d_ssd1320_cs1_160x132_nhd_powersave0_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x0af),		                /* ssd1320: display on */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

static const uint8_t u8x8_d_ssd1320_cs1_160x132_nhd_powersave1_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x0ae),		                /* ssd1320: display off */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

/*
  input:32
    one tile (8 Bytes; 1 byte per column)
  output:
    Tile for SSD1320 (32 Bytes)

  The origin of the display seems to be in the upper right-hand corner. Therefore
  compared to SSD1322, the order inside each byte is swapped.
*/

static uint8_t u8x8_ssd1320_to32_dest_buf[32];
[... 212 lines cut ...]
}
```

The entry in u8g2_d_setup.c looks like this:
```
/* ssd1320 f */
void u8g2_Setup_ssd1320_cs1_160x132_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
{
  uint8_t tile_buf_height;
  uint8_t *buf;
  u8g2_SetupDisplay(u8g2, u8x8_d_ssd1320_cs1_160x132, u8x8_cad_001, byte_cb, gpio_and_delay_cb);
  buf = u8g2_m_20_16_f(&tile_buf_height);
  u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
}
```

But when I draw a simple frame you can see the image is cropped:
```
    u8g2.clearBuffer();
    u8g2.drawFrame(2, 2, 156, 128);
    u8g2.sendBuffer();
```
[image]

Any thoughts @olikraus ? Could this still be a setup issue?

olikraus, 2021-03-07:
Please excause for not being present for the last weeks. Unfortunately there is currently not much time left to work on u8g2.
Also thanks a lot to @ddlqn for your reply.

olikraus, 2021-03-07:
I tookover the (updated?) code from @buildxyz-git into the repo. Will do some updates also: I first changed the dev fn name to `u8x8_d_ssd1320_160x132` and removed the cs1 (I think you could just use two different constructors with different cs lines)

olikraus, 2021-03-07:
The tile height has to be 17 because tile_height * 8 has to be larger or equal to the pixel_height. Fixed this in the repo.

olikraus, 2021-03-07:
I have created beta release 2.29.1

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Can we have a look at the u8x8 constructors?

//U8X8_SSD1320_160X132_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
//U8X8_SSD1320_160X132_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  

Please run and execude the u8x8 FlipMode example (https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8x8/FlipMode/FlipMode.ino)
You can uncomment one constructor from above and update the pin numbers to your local hardware setup.

What will you see? Can you sent a picture? The example will flip between two pictures, are both valid?

buildxyz-git, 2021-03-07:
Thanks you for making the time @olikraus !

I installed the U8g2 beta and ran the u8x8 flip mode example as instructed. Unfortunately I got junk:
https://user-images.githubusercontent.com/16586332/110248860-f8d39500-7f27-11eb-94b3-5125c01d0681.MOV

I saw the u8x8_d_ssd1320.c and did diff against my "working" version and they are the same. I have once I also updated the tile_height = 17 with no change in my almost working example shown in earlier pictures.

So I'm really not sure why I got such odd results with your updated library.  I ran my old example which has the display cut off at a height of 30 pixels and it works the same as id did before.
[image]

olikraus, 2021-03-07:
hmm... strange. How will a u8g2 example look like? For example the u8g2 page buffer / full buffer Hello World?

buildxyz-git, 2021-03-07:
It's weird behavior for sure. How about the simple Hello World example to demonstrate the issue? 

```
#include <Arduino.h>
#include <U8g2lib.h>

U8G2_SSD1320_CS1_160X132_F_4W_HW_SPI u8g2(U8G2_R0, 32, 31, 30);

void setup()
{
    u8g2.setFont(u8g2_font_6x10_tf);
    u8g2.begin();
}

void loop()
{
    u8g2.clearBuffer();

    u8g2.drawStr(0,u8g2.getMaxCharHeight() * 1, "Hello World! 1");
    u8g2.drawStr(0,u8g2.getMaxCharHeight() * 2, "Hello World! 2");
    u8g2.drawStr(0,u8g2.getMaxCharHeight() * 3, "Hello World! 3");
    u8g2.drawStr(0,u8g2.getMaxCharHeight() * 4, "Hello World! 4");
    u8g2.drawStr(0,u8g2.getMaxCharHeight() * 5, "Hello World! 5");

    u8g2.sendBuffer();
    delay(100);
}
```
This is running in full buffer mode I presume according how the constructor is being called U8g2lib.h:

**U8g2lib.h**
```
...
class U8G2_SSD1320_CS1_160X132_F_4W_HW_SPI : public U8G2 {
  public: U8G2_SSD1320_CS1_160X132_F_4W_HW_SPI(const u8g2_cb_t *rotation, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE) : U8G2() {
    u8g2_Setup_ssd1320_cs1_160x132_f(&u8g2, rotation, u8x8_byte_arduino_hw_spi, u8x8_gpio_and_delay_arduino);
    u8x8_SetPin_4Wire_HW_SPI(getU8x8(), cs, dc, reset);
  }
};
...
```

The "u8g2_m_20_16_f" buffer may not be correct as well?
**src/clib/u8g2_d_setup.c**
```
/* ssd1320 f */
void u8g2_Setup_ssd1320_cs1_160x132_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
{
  uint8_t tile_buf_height;
  uint8_t *buf;
  u8g2_SetupDisplay(u8g2, u8x8_d_ssd1320_cs1_160x132, u8x8_cad_001, byte_cb, gpio_and_delay_cb);
  buf = u8g2_m_20_16_f(&tile_buf_height);
  u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
}
```

[image]

olikraus, 2021-03-07:
Hmm, ok the "u8g2_m_20_16_f" looks very strange.
The hello world output is from my u8g2 beta version? So it seems ok, except for the missing lower part?

olikraus, 2021-03-07:
The setup looks correct in the repo:
https://github.com/olikraus/u8g2/blob/7fc361f3d2ab218289114e9a6631e40b1143e773/csrc/u8g2_d_setup.c#L5550-L5556

Actually you have used the previous CS1 constructor. Is there any u8g2 picture of the u8g2 beta?

buildxyz-git, 2021-03-07:
In the previous hello world example I was reverting back to the latest working u8g2 I had modified so I was not using the beta, I just wanted to demonstrate the closest thing to a working setup. 
So yes, it appears just the bottom of the buffer is cropped at 30 px high.
I know you recommend instantiating two of the same constructors for the two controllers but I seem to need write different initialization registers in the setup routine for each controller. The two controllers for one screen is probably a “solve later” problem at this point.
I’ll try and update the buffer to your beta code here shortly.

olikraus, 2021-03-07:
> I’ll try and update the buffer to your beta code here shortly.

It would be really cool if you could test the u8g2 beta with the u8g2 hello world example.

buildxyz-git, 2021-03-07:
I sure can. Helloworld example sketch. You can see it prints the string but it's scrolling and glitching for some reasone.

```
U8G2_SSD1320_160X132_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 32, /* dc=*/ 31, /* reset=*/ 30); 

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}
```

https://user-images.githubusercontent.com/16586332/110259112-e9217400-7f5a-11eb-8057-e8e5486fef8d.MOV

buildxyz-git, 2021-03-08:
Changing the following  from **u8x8_cad_011** to **u8x8_cad_001** in _u8g2_Setup_ssd1320_160x132_f()_ from _u8g2_d_setup.c_ seems to make the hello world in your beta release function.

I still get the height cropping issue however.

```
/* ssd1320 f */
void u8g2_Setup_ssd1320_160x132_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
{
  uint8_t tile_buf_height;
  uint8_t *buf;
  u8g2_SetupDisplay(u8g2, u8x8_d_ssd1320_160x132, **u8x8_cad_001**, byte_cb, gpio_and_delay_cb);
  buf = u8g2_m_20_17_f(&tile_buf_height);
  u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
}
```

The slightly modified helloworld is here:
```
U8G2_SSD1320_160X132_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 32, /* dc=*/ 31, /* reset=*/ 30); 

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.drawStr(0,20,"Hello World!");  // write something to the internal memory
  u8g2.drawStr(0,30,"Hello World!");  // write something to the internal memory
  u8g2.drawStr(0,40,"Hello World!");  // write something to the internal memory
  u8g2.drawStr(0,50,"Hello World!");  // write something to the internal memory
  u8g2.drawStr(0,60,"Hello World!");  // write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}
```
The results:
[image]

olikraus, 2021-03-08:
> Changing the following from u8x8_cad_011 to u8x8_cad_001 

Wow, thanks a lot for pointing this out.

olikraus, 2021-03-08:
I think the missing lines are due to the setup for the 160x32 OLED, the a8 cmd has to be modified. According to your documentation (section 4 on page 12) we have to use 0x83.

olikraus, 2021-03-08:
I have created beta 2.29.2
I also updated some commands of the init sequence. I changed the d3 command according to your documentation, but I am not sure whether this is really correct.
The change for the LEFT DISPLAY (default display) is here:
https://github.com/olikraus/u8g2/blob/c65286667efc6a465eadfcc0e79136369b41172a/csrc/u8x8_d_ssd1320.c#L393

Additionally I have added the constructor for the 160x32 OLED, but I guess this is more of interest for @ddlqn 

Can you download 2.29.2 and check your left display?

buildxyz-git, 2021-03-08:
Nailed it!!!

[image]

Both lines are required it seems:
```
U8X8_CA(0xd3, 0x0e),        /* CS1 */
U8X8_CA(0xa8, 0x83),      /* multiplex ratio 1/132 Duty  */ 
```
I also had to bump up the contrast it seems now:
`U8X8_CA(0x81, 0xFF),      /* contrast                                 */ `

Any tips on how to most efficiently use two controllers/constructors? I'd like to end this thread with 320px wide image!

olikraus, 2021-03-09:
> I also had to bump up the contrast it seems now:
> U8X8_CA(0x81, 0xFF), /* contrast */ 

I think we need to apply the other settings from your datasheet also.

> I'd like to end this thread with 320px wide image!

Sure, as a next step: Can you execute the u8g2 flip mode example. There should be a border frame in all 8 modes which are displayed. Can you check this?

buildxyz-git, 2021-03-09:
Sure can!

U8G2 flipmode example results:
https://user-images.githubusercontent.com/16586332/110498831-87711f00-80ac-11eb-8d9f-f7233b4fca01.MOV

U8X8 Flipmode example results:
https://user-images.githubusercontent.com/16586332/110498896-948e0e00-80ac-11eb-970d-35c634cdd478.MOV

olikraus, 2021-03-09:
u8g2 looks great, right? u8x8 might not work here, which is ok.

> Any tips on how to most efficiently use two controllers/constructors? I'd like to end this thread with 320px wide image!

First of all you need to enable 16 bit mode (uncomment #define in u8g2.h, see FAQ)

Then you need to create two constructors, one for the left and another for the right side, let's say u8g2_0 and u8g2_1
u8g2_0 requires flip mode 0
u8g2_1 requires flip mode 1
(https://github.com/olikraus/u8g2/wiki/u8g2reference#setflipmode)

Then you need a current u8g2 pointer:
U8G2 *u8g2_current;
A offset variable might also be usefull (it will be either 0 or 160):
u8g2_uint_t offset;

a procedure might be usefull to set either left or right side:
```
void setLeftSide() {
  u8g2_current = &u8g2_0;
  offset = 0;
}
void setRightSide() {
  u8g2_current = &u8g2_1;
  offset = 160;
}
```

You probably need a new draw functions. Lets consider:
`u8g2_uint_t U8g2::drawStr(u8g2_uint_t x, u8g2_uint_t y, const char *s)  `

For this you need:
```
void drawStr(u8g2_uint_t x, u8g2_uint_t y, const char *s) {
  u8g2_current->drawStr(x-offset,y,s)
}
```

For each screen, you should create a procedure. Lets assume a welcome screen:
```
void drawWelcome() {
  u8g2_current->setFont(<some font>);
  drawStr(0, 20, "Welcome");
}
```

If you use page buffer mode, you need to execute the loop twice:
```
  setLeftSide()
  u8g2_current->firstPage();
  do {
    drawWelcome();
  } while (  u8g2_current->nextPage() );
  setRightSide()
  u8g2_current->firstPage();
  do {
    drawWelcome();
  } while (  u8g2_current->nextPage() );
```

olikraus, 2021-03-10:
Good idea to send the code. There is a small issue with the constructors. From the u8g2 FAQ:
https://github.com/olikraus/u8g2/blob/c65286667efc6a465eadfcc0e79136369b41172a/doc/faq.txt#L258-L260

So it should be:
```
U8G2_SSD1320_160X132_2_4W_HW_SPI u8g2_0(U8G2_R0, /* cs=*/ 26, /* dc=*/ OLED_D1_DC, /* reset=*/ OLED_D1_RST);
U8G2_SSD1320_160X132_2_4W_HW_SPI u8g2_1(U8G2_R0, /* cs=*/ 27, /* dc=*/ OLED_D1_DC, /* reset=*/ U8X8_PIN_NONE);
```

olikraus, 2021-03-10:
Actually I am not so happy with your contrast value of 255. The value should be lesser, which means, that the init sequence is not yet correct. In fact, I have only modified the geometry setup for your display, but not the electronics part. Instead the charge pump and volume setup is still from the 160x32 display. 
Let me know whether you want to go ahead with fine tuning the display setup.

buildxyz-git, 2021-03-14:
Check out the full screen animation [HERE!](https://www.instagram.com/p/CMXS-LQAJnl)!

There are some "tearing" effects when animating both sides of the screen as one side is updated before the other. Also there are some other small artifacts like dim scanning lines when the contrast is set to higher values.

I took a stab at including all the other initialization parameters to follow the examples in the datasheet but nothing really changed. It looks like this now:

```
static const uint8_t u8x8_d_ssd1320_cs1_160x132_init_seq[] = {
    
    U8X8_DLY(1),
    U8X8_START_TRANSFER(),    /* enable chip, delay is part of the transfer start */
    U8X8_DLY(1),
    
    U8X8_C(0xae),		          /* display off */
    U8X8_CA(0xfd, 0x10),      /* Set command lock */
    U8X8_CA(0x20, 0x00),	    /* Memory Addressing Mode: Horizontal */
    U8X8_CA(0xa2, 0x00),			/* display start line */
    U8X8_CA(0xd3, 0x0e),      /* Set display offset */

    U8X8_CA(0xd5, 0xc2),			/* set display clock divide ratio/oscillator frequency (set clock as 80 frames/sec) */  
    U8X8_CA(0xa8, 0x83),			/* multiplex ratio 1/132 Duty */  

    U8X8_C(0xa0),	            /* Set Segment Re-Map: column address 0 mapped to SEG0 */ 
    U8X8_C(0xc8),	            /* Set COM Output Scan Direction: normal mode */
    U8X8_C(0xbf),             /* Linear LUT */	
    
    U8X8_CA(0xda, 0x12),	    /* Set SEG Pins Hardware Configuration */   
    U8X8_CA(0xad, 0x10),			/* Internal IREF Enable */
    U8X8_CA(0xbc, 0x1e),			/* Precharge voltage */
    U8X8_CA(0xd9, 0x72),			/* Precharge period */
    U8X8_CA(0xbd, 0x03),      /* Set VP */
    U8X8_CA(0xdb, 0x30),      /* Set VCOMH */

    U8X8_CA(0x81, 0x7F),			  /* Contrast */  
    U8X8_C(0xa4),		        	  /* display on */  
    U8X8_C(0xa6),		            /* normal display */

    U8X8_DLY(1),					      /* delay 2ms */

    U8X8_END_TRANSFER(),        /* disable chip */
    U8X8_END()             			/* end of sequence */
};
```

olikraus, 2021-03-14:
I have created a new beta 2.29.3 with the new init sequence. It should be brighter then before (hopefully). Can you test the same?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

buildxyz-git, 2021-03-14:
Oh we just commented at the same time! Let me check out 2.29.3!

buildxyz-git, 2021-03-14:
Hmmm, I ran the u8g2 HelloWorld and didn't notice a difference. You did update the MUX ratio, I'm not even sure what that does. What are normally the registers which control brightness and refresh rates?

olikraus, 2021-03-14:
> Hmmm, I ran the u8g2 HelloWorld and didn't notice a difference. You did update the MUX ratio, I'm not even sure what that does. What are normally the registers which control brightness and refresh rates?

Actually I just took over values from the display vendor (I think you posted the datasheet with the init sequence once).
Will your init sequence work better?

olikraus, 2021-09-02:
I think it's time to close this issue. Let me know if there is something else todo.

closed 2021-09-02 by olikraus

buildxyz-git, 2021-09-02:
All good! Close the issue!

BiatuAutMiahn, 2022-06-24:
What about implementing the bull buffer in the library itself? instead of flipping through both sides?

olikraus, 2022-06-24:
Technically I do not see an advantage of doing this in u8g2 itself. Codesize and RAM increase will be there either it is part of u8g2 or not.

steiraa, 2024-01-12:
> Using.two displays as one will be simple. Just draw everything on both displays with suitable offset. U8g2 will do the clipping. I worry more about the setup.

Hello, may I ask how this would work? I have this exact same üroblem with [this ](https://www.buydisplay.com/catalog/product/view/id/1847/s/arduino-raspberry-pi-3-83-inch-oled-module-320x132-spi-16-level-grayscale/category/242/)display